bitdump/BLHeli #135: 500Hz PWM Input and start point

Jimqjp, 2016-04-05:
I am working on a project of brushless update of Walkera Master CP.

The receiver of Master CP outputs 500 Hz PWM for both main and tail motors.  

The first step is to upgrade the main motor. I am using a SuperSimple 30A ESC with V14.2.  

Though the manual states that the lowest PWM frequency supported is 1 kHz, it seems working with the ESC. However, instead of 0 to 100% PWM working range, the ESC accepts PWM from about 55% to 100%.  Is this supposed to be, or because of the half of the PWM frequency? 

The main motor can work with this, but it will not work with tail motor.  The tail locking loop assumes 0 to %100 for 0 to 100% rpm.

Is there a way to fully utilize the PWM duty?

Thanks in advance.

sskaug, 2016-04-06:
It sounds like the ESC detects your input signal as 1-2ms regular RC signal. Which is the intended way of detecting a 500Hz input signal. You will need to change the code if you are to use a 500Hz input signal

Jimqjp, 2016-04-06:
Thank you for your reply.

For the latest version 14.4, based on 6 searched "RCP_PWM_FREQ_1KHZ" I will do the following modifications.

Do you mind to point out where I need to modify other than these? 

I appreciate your help very much.
 
-----------------------------------------------------------------------------------------------------------------------------------
1. Line 1111: Add a 500Hz PWM .EQU before 1Khz, and shift rest numbers by one 

.EQU	RCP_PWM_FREQ_0_5KHZ		= 	0		; RC pulse pwm frequency is 0.5kHz
.EQU	RCP_PWM_FREQ_1KHZ		= 	1		; RC pulse pwm frequency is 1kHz
.EQU	RCP_PWM_FREQ_2KHZ		= 	2		; RC pulse pwm frequency is 2kHz
.EQU	RCP_PWM_FREQ_4KHZ		= 	3		; RC pulse pwm frequency is 4kHz
.EQU	RCP_PWM_FREQ_8KHZ		= 	4		; RC pulse pwm frequency is 8kHz
.EQU	RCP_PWM_FREQ_12KHZ		= 	5		; RC pulse pwm frequency is 12kHz
.EQU	PGM_DIR_REV			= 	6		    ; Programmed direction. 0=normal, 1=reversed
.EQU	PGM_RCP_PWM_POL		=	7		    ; Programmed RC pulse pwm polarity. 0=positive, 1=negative
.EQU	FULL_THROTTLE_RANGE		= 	8		; When set full throttle range is used (1000-2000us) and stored calibration values are ignored

2. Line 2279:
rcp_int_check_1kHz:
	; Check if pwm frequency is 1kHz
	cpi	I_Temp1, low(2200)			; If below 1100us, 1kHz pwm is assumed
	ldi	XL, high(2200)
	cpc	I_Temp2, XL
	brcc	rcp_int_check_0_5kHz

	ldi	XL, (1<<RCP_PWM_FREQ_1KHZ)
	mov	I_Temp4, XL
	ldi	XL, 120					; Set period tolerance requirement (LSB)
	mov	I_Temp3, XL
	rjmp	rcp_int_restore_edge
    
rcp_int_check_0_5kHz:
	; Check if pwm frequency is 0.5kHz
	cpi	I_Temp1, low(4400)			; If below 2200us, 0.5kHz pwm is assumed
	ldi	XL, high(4400)
	cpc	I_Temp2, XL
	brcc	rcp_int_restore_edge

	ldi	XL, (1<<RCP_PWM_FREQ_0_5KHZ)
	mov	I_Temp4, XL
	ldi	XL, 240					; Set period tolerance requirement (LSB)
	mov	I_Temp3, XL
    
rcp_int_restore_edge:

3. Line 2363
	sbrc	Flags3, RCP_PWM_FREQ_1KHZ		; Is RC input pwm frequency 1kHz?
	rjmp	rcp_int_pwm_divide				; Yes - branch forward
    
	lsr	I_Temp2						        ; No - 1kHz. Divide by 2 again
	ror	I_Temp1
    
	sbrc	Flags3, RCP_PWM_FREQ_0_5KHZ		; Is RC input pwm frequency 0.5kHz?
    
4. Line 2652:
	ldi	XL, ((1<<RCP_PWM_FREQ_0_5KHZ)+(1<<RCP_PWM_FREQ_1KHZ)+(1<<RCP_PWM_FREQ_2KHZ)+(1<<RCP_PWM_FREQ_4KHZ)+(1<<RCP_PWM_FREQ_8KHZ)+(1<<RCP_PWM_FREQ_12KHZ))

5. Line 5639:
    addi XL, ((1<<RCP_PWM_FREQ_0_5KHZ)+(1<<RCP_PWM_FREQ_1KHZ)+(1<<RCP_PWM_FREQ_2KHZ)+(1<<RCP_PWM_FREQ_4KHZ)+(1<<RCP_PWM_FREQ_8KHZ)+(1<<RCP_PWM_FREQ_12KHZ))        
    
6. Line 5661:
    andi XH, !((1<<RCP_PWM_FREQ_0_5KHZ)+(1<<RCP_PWM_FREQ_1KHZ)+(1<<RCP_PWM_FREQ_2KHZ)+(1<<RCP_PWM_FREQ_4KHZ)+(1<<RCP_PWM_FREQ_8KHZ)+(1<<RCP_PWM_FREQ_12KHZ))

Jimqjp, 2016-04-14:
I successfully modified the code and my ESC accepts 500Hz PWM signal now! 

The table size is limited by 8 extra value can cause compile failure. So in stead of adding another value, the table is shifted, the 12KHz is removed from the bottom and 500Hz is added to the top (0). 

May be because the 500Hz pulse width is close to the pulse width of  PPM (about 1ms to 2ms), once PWM is turned one, PPM signal cannot recognized. I do not expect to use both with the same setup anyway.

kiwi-craig, 2016-06-12:
Hi, I 'm converting my master, any chance I could get a copy of your file ?  Many Thanks

crainic@ihug.co.nz

Jimqjp, 2016-06-12:
Attached is the asm file. I was using Atmel Stduio 7 to build.

`;**** **** **** **** ****
;
; BLHeli program for controlling brushless motors in helicopters and multirotors
;
; Copyright 2011, 2012 Steffen Skaug
; This program is distributed under the terms of the GNU General Public License
;
; This file is part of BLHeli.
;
; BLHeli is free software: you can redistribute it and/or modify
; it under the terms of the GNU General Public License as published by
; the Free Software Foundation, either version 3 of the License, or
; (at your option) any later version.
;
; BLHeli is distributed in the hope that it will be useful,
; but WITHOUT ANY WARRANTY; without even the implied warranty of
; MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
; GNU General Public License for more details.
;
; You should have received a copy of the GNU General Public License
; along with BLHeli.  If not, see <http://www.gnu.org/licenses/>.
;
;**** **** **** **** ****
;
; The software was initially designed for use with Eflite mCP X, but is now adapted to copters/planes in general
;
; The software was inspired by and started from from Bernard Konze's BLMC: http://home.versanet.de/~bkonze/blc_6a/blc_6a.htm
; And also Simon Kirby's TGY: https://github.com/sim-/tgy
;
; This file is best viewed with tab width set to 5
;
; The input signal can be positive 1kHz, 2kHz, 4kHz, 8kHz or 12kHz PWM (e.g. taken from the "resistor tap" on mCPx)
; And the input signal can be PPM (1-2ms) or OneShot125 (125-250us) at rates up to several hundred Hz.
; The code autodetects the various input modes/frequencies
; The code can also be programmed to accept inverted input signal.
;
; The first lines of the software must be modified according to the chosen environment:
; Uncomment the selected ESC and main/tail/multi mode
; BESC EQU "ESC"_"mode" 						
; 
;**** **** **** **** ****
; Revision history:
; - Rev0.0: Initial revision
; - Rev1.0: Governor functionality added
; - Rev1.1: Increased tail gain to 1.0625. Implemented for tail only
;		  Decreased governor proportional and integral gain by 4
; 		  Fixed bug that caused tail power not always to be max
; - Rev1.2: Governor integral gain should be higher in order to achieve full PWM range
;           Integral gain can be higher, and is increased by 2x. An integral of +-128 can now be added to requested PWM
; - Rev1.3: Governor integral extended to 24bit
;		  Governor proportional gain increased by 2x	
;		  Added slow spoolup/down for governor
;		  Set pwm to 100% (do not turn off nFET) for high values of current pwm
;		  Added support for PPM input (1us to 2us pulse)
;		  Removed USE_COMP_STORED as it was never used
; - Rev2.0  Added measurement of pwm frequency and support for 1kHz, 2kHz, 4kHz and 8kHz
;           Optimized pwm on and off routines
;           Improved mosfet switching in beep routines, to reduce current draw
;           Added support for ICP1 interrupt pin input
;           Added ADC measurement of supply voltage, with limiting of main motor power for low voltage
;           Miscellaneous other changes
; - Rev2.1  Rewritten INT0 routine to be similar to ICP
;           Reduced validation threshold (RCP_VALIDATE)
;           Removed requirement for RCP to go to zero again in tail arming sequence
;           Removed PPM support
; - Rev2.2  Added support for HC 5A 1S ESC with Atmega48V MPU
;           Increased governor proportional gain by 2x
; - Rev3.0  Added functionality for programming from TX
[... 6075 lines cut ...]
	cp	XH, Temp2					; Is counter above requirement?
	brcs	start_check_rcp			; No - proceed

	cbr	Flags1, (1<<STARTUP_PHASE)	; Clear startup phase flag
	sbr	Flags1, (1<<INITIAL_RUN_PHASE); Set initial run phase flag
	sts	Initial_Run_Rot_Cnt, Temp3	; Set initial run rotation count
.IF MODE == 1	; Tail
	ldi	XH, 0xFF
	sts	Pwm_Limit, XH				; Allow full power
.ENDIF
.IF MODE == 2	; Multi
	lds	XH, Pwm_Spoolup_Beg
	sts	Pwm_Limit, XH
	sts	Pwm_Limit_By_Rpm, XH
.ENDIF
	rjmp	normal_run_checks

start_check_rcp:
	lds	XH, New_Rcp				; Load new pulse value
	cpi	XH, RCP_STOP				; Check if pulse is below stop value
	brcs	PC+2

	rjmp	run1						; Continue to run 

	rjmp	run_to_wait_for_power_on

kiwi-craig, 2016-06-12:
thanks, unfortunately coding is beyond capabilities with my setup at home (imac)   but I can flash BLheli files from an old laptop (XP).  

if you could send me your revised file to my home email I'd be grateful, there are allot of guys on RC Groups - Master CP thread who would appreciate this

crainic@ihug.co.nz

Many Thanks

Jimqjp, 2016-06-13:
I am using SuperSimple 30A ESC. The complied .hex file is only good for that type. Need to build separately for each compatible ESC.

What ESC are you going to use?  Maybe I can help to build the type for your ESC.

kiwi-craig, 2016-06-13:
Jimgjp 

okay, right. 

I'm running with the DYS BL30A, same file
used as the DYS BL20A

http://www.banggood.com/DYS-BL30A-Mini-30A-BLHeli-ESC-OPTO-2-6s-for-RC-Multicopter-p-975799.html?cur_warehouse=CN

The DYS 20 / 30 units seem decent, and compact as well. 

Many Thanks

Craig 

> I am using SuperSimple
30A ESC. The complied .hex file is only good for that type. Need to
build separately for each compatible ESC. 
> 
> What ESC are you going
to use? Maybe I can help to build the type for your ESC. 
> 
> You
are receiving this because you commented.
> Reply to this email
directly, view it on GitHub [1], or mute the thread [2].

Links:
------
[1]
https://github.com/bitdump/BLHeli/issues/135#issuecomment-225480022
[2]
https://github.com/

kiwi-craig, 2016-06-14:
sorry for the confusion above
DYS BL and or Afro esc seem to be the pick.   

thanks

Craig

Jimqjp, 2016-06-22:
DYS uses SiLabs chip, which is 8051 family. The mod is for Atmega family only.

otrofox, 2017-05-24:
Hi,

Thanks so much for your work.
It has inspired me to try a similar mod for my Syma X8HC, which works at 6Khz PWM.

I'm using SiLabs, but I think the code to change is the same as for Atmel chips.

I'll try to change 8Khzs code for 6Khz parameters, instead lf adding a new frequency mode.
Can you please give me any advice?
As far as I know, the only change to do should be:

pca_int_check_8kHz:
	; Check if pwm frequency is 8kHz
	clr	C
	mov	A, Temp1
	subb	A, #low(**# 480**)				; If below **# 240us, 6kHz** pwm is assumed
	mov	A, Temp2
	subb	A, #high(**# 480**)
	jnc	pca_int_check_4kHz

	clr	A
	setb	ACC.RCP_PWM_FREQ_8KHZ
	mov	Temp4, A
	mov	Temp3, #**# 20**				; Set period tolerance requirement (LSB)
	ajmp	pca_int_restore_edge_set_msb

Thanks for your help!!!