bitdump/BLHeli #146: Support for SiLabs F330 LittleDevil 20a ESC

AldoPadilla, 2016-05-11:
Arduino connect not working. This ESC is not in the supported list.

http://www.readytoflyquads.com/20a-littledevil-esc

le3dmax, 2016-05-11:
I thought it's just a littlebee 20a copy...

AldoPadilla, 2016-05-11:
I thought so too. I connected it to my Arduino for One-Wire programming, as I do with my other ESCs, and no luck. Tested it with four others of the same. 

Other ESCs connect and program just fine. 

Mac OS + Wine + Arduino Uno + Latest BLHeli

sskaug, 2016-05-11:
Maybe a pulldown resistor on the signal input?

AldoPadilla, 2016-05-11:
Thanks for the advice. Can you elaborate? How should I perform this test?

le3dmax, 2016-05-11:
Did you try to connect the Blheli Suite through your FC? betaflight 2.6.1
works like a charm!

AldoPadilla, 2016-05-11:
No. No passthrough on my Motolab Tornado. Connecting direct to ESC via Arduino Uno.

AldoPadilla, 2016-05-11:
@sskaug 
A resistor between the signal wire and ground? 
What value?

4712, 2016-05-11:
I guess the problem is more around the Arduino handling...
Please describe exactly the steps you took and post the log.tlg.

AldoPadilla, 2016-05-11:
Sure. I pretty much followed the instruction PDF 'BLHeli Programming adapters' section 'Arduinos as USB/UART 1-Wire bridge'

- Under 'Make Interfaces' I flashed the Arduino for 'Arduino BLHeli Bootloader (C 1)' under the appropriate COM port. Got validation of success.
- Selected 'C: SILABS BLHELI Bootloader (USB/COM)' under 'Interfaces' 
- Connected Arduino ground to the ESC signal and ground. Arduino ground to black. Arduino Uno Pin 0 to white. 
- Selected the appropriate COM port and hit connect. 
- Then powered up the ESC

This process has worked with my other ESC's just fine. 

Since it didn't with this particular ESC I attempted different orders of power-ups and hitting 'Connect'.
Still no luck. 

I'm not at home, I'll upload the log.tlg tonight

4712, 2016-05-11:
@AldoPadilla Uno Pin 0 is wrong. Should be 11.

AldoPadilla, 2016-05-11:
@4712 
Interesting! I didn't try that for sure.
I'm curious now... Why does pin 0 work with my other ESCs?

4712, 2016-05-11:
Yes very Interesting, I never intended this, but it works... Never tried before. ;)

AldoPadilla, 2016-05-11:
@4712 Thanks, I'll give it a shot tonight and report back.

4712, 2016-05-11:
@AldoPadilla You will see, it works with digital pin 11.
Although I could get it working using pin 0 (tx) with the Aikon SEFM 30A, it dos not work with the LB 20A.
[image]
[image]

AldoPadilla, 2016-05-11:
Wow!
Thanks for the thorough help on this. Can't wait to try it out later.

AldoPadilla, 2016-05-11:
@4712 
Awesome! Worked on the first try. 
Thanks again!

closed 2016-05-22 by sskaug