bitdump/BLHeli #151: BLHeli_s.asm <> B.inc

JerryXing, 2016-05-25:
BLHeli_s.asm
B_			EQU 2	; X  X  RC X  MC MB MA CC    X  X  Ac Ap Bc Bp Cc Cp

B.Inc
;*********************
; PORT 1 definitions *
;*********************
;			EQU	7	;i
;			EQU	6	;i
ApwmFET		EQU	5	;o
AcomFET		EQU	4	;o
BpwmFET		EQU	3	;o
BcomFET		EQU	2	;o
CpwmFET		EQU	1	;o
CcomFET		EQU	0	;o

JerryXing, 2016-05-25:
B.Inc
; Hardware definition file "B". Equals "A", but with A and C fets swapped
; X  X  RC X  MC MB MA CC    X  X  Ac Ap Bc Bp Cc Cp

sskaug, 2016-05-25:
Great find. Now updated to:
B.inc: 
; Hardware definition file "B". Equals "A", but with A and C fets swapped and pwm and com fets swapped
; X  X  RC X  MC MB MA CC    X  X  Ap Ac Bp Bc Cp Cc

B.asm:
B_			EQU 2	; X  X  RC X  MC MB MA CC    X  X  Ap Ac Bp Bc Cp Cc

This change is only done locally for now, I'll commit it later.

closed 2016-07-11 by JerryXing