bitdump/BLHeli #152: Issue with firmware 14.5 on DYS BL20A

jszewczyk, 2016-05-26:
Hey.

I have drone 250 with ESC DYS BL20A BLHeli and flight controller naze32 ver6 and tarot MT2204 CCW 2300KV engines (medium timing). If I use firmware 14.3, 14.4, all is fine. If I upload 14.5 my engines have problem with synchronization.

Thanks for help. Jacek

sskaug, 2016-05-26:
How does it run with rev 14.6?

jszewczyk, 2016-05-26:
In my BLHeliSuite is available only 14.5 from newer firmwares. Where can I find new firmware version compatible with my ESC?

deepimpactio, 2016-05-26:
Right here on GitHub if you have Silabs here is link
https://github.com/bitdump/BLHeli/tree/master/SiLabs

deepimpactio, 2016-05-26:
Sorry here are hex
https://github.com/bitdump/BLHeli/tree/master/SiLabs/Hex%20files

4712, 2016-05-26:
http://www.mediafire.com/download/c1m6tvdpfmfsv8j/patch_16.1_14.6_for_BLHeliSuite16.0.14.5.0.1.zip

jszewczyk, 2016-05-26:
Thanks for patch :)