bitdump/BLHeli #154: 16.X ADC Bug

JerryXing, 2016-05-30:
mov	REF0CN, #2Ch	; Set vdd (3.3V) as reference. Enable temp sensor and bias

Use AGND;

But

5 GNDSL 0 RW Analog Ground Reference.
Selects the ADC0 ground reference.
Value Name Description
0 GND_PIN The ADC0 ground reference is the GND pin.
1 AGND_PIN The ADC0 ground reference is the P0.1/AGND pin.

BLHeli_S.asm
; List of enumerated supported ESCs
A_			EQU 1	; X  X  RC X  MC MB MA CC    X  X  Cc Cp Bc Bp Ac Ap
B_			EQU 2	; X  X  RC X  MC MB MA CC    X  X  Ac Ap Bc Bp Cc Cp
C_			EQU 3	; Ac Ap MC MB MA CC X  RC    X  X  X  X  Cc Cp Bc Bp
D_			EQU 4	; X  X  RC X  CC MA MC MB    X  X  Cc Cp Bc Bp Ac Ap	Com fets inverted
E_			EQU 5	; L1 L0 RC X  MC MB MA CC    X  L2 Cc Cp Bc Bp Ac Ap	A with LEDs
F_			EQU 6	; X  X  RC X  MA MB MC CC    X  X  Cc Cp Bc Bp Ac Ap
G_			EQU 7	; X  X  RC X  CC MA MC MB    X  X  Cc Cp Bc Bp Ac Ap	Like D, but noninverted com fets

My email:bsxing@gmail.com

sskaug, 2016-05-30:
Thanks for the find! This is indeed a bug, proper GND should be selected for the ADC, so it should be "mov REF0CN, #0Ch". The effect of this is as far as I can see that the temperature limiting does not work properly (ADC negative input is not ground, but some higher voltage).

JerryXing, 2016-05-30:
I agree with you.
Temperature may be a new bug.
or
Initialize_Adc MACRO
	mov	REF0CN, #0Ch	; Set vdd (3.3V) as reference. Enable temp sensor and bias
IF MCU_48MHZ == 0
	mov	ADC0CF, #5Dh		; ADC clock 2MHz
ELSE
	mov	ADC0CF, #0C5h		; ADC clock 2MHz
ENDIF
	mov	ADC0MX, #10h	; Select temp sensor input
	mov	ADC0CN0, #80h	; ADC enabled 
	mov	ADC0CN1, #01h	; Common mode buffer enabled
ENDM

Read_Adc_Result MACRO
	mov	Temp1, ADC0H
	mov	Temp2, #0
ENDM

sskaug, 2016-05-30:
Right - the PGA gain must also be set to 1.

sskaug, 2016-05-30:
10bit conversion still fits the FW ported from base BLHeli better.

JerryXing, 2016-05-30:
I see. Good code.

closed 2016-07-11 by JerryXing