bitdump/BLHeli #156: EFM8BB1x==C8051F850

JerryXing, 2016-05-30:
Device SiLabs EFM8BB1x/2x/C8051F850 ?
Is ok?

sskaug, 2016-05-30:
As far as I know, F850 and EFM8BB1 are compatible (or equal)

JerryXing, 2016-05-30:
yes.
BB1 reg == 850reg
My Test is ok.
Doc add F850?

sskaug, 2016-05-31:
My tests has shown some differences between BB1 and F850. Although the devices are compatible, I think. But why would anyone use F850? And if they do, why not just treat it (from FW) like a BB1.

JerryXing, 2016-05-31:
What are the differences?
I have a lot of F850.
But BB1 will wait for 2 months.

sskaug, 2016-05-31:
I found some subtle differences in the hardware pwm generation circuit, where I think the BB1 is more robust. But still, AFAIK, the f850 should work.

JerryXing, 2016-05-31:
I did not see the difference

BB1 Center Aligned PWM
When configured for center-aligned mode, a module generates an edge transition at two points for every 2(N+1) PCA clock cycles,
where N is the selected PWM resolution in bits. In center-aligned mode, these two edges are referred to as the “up” and “down” edges.
The polarity at the output pin is selectable and can be inverted by setting the appropriate channel bit to 1 in the PCA0POL register.
The generated waveforms are centered about the points where the lower N bits of the PCA0 counter are zero. The (N+1)th bit in the
PCA0 counter acts as a selection between up and down edges. In 16-bit mode, a special 17th bit is implemented internally for this
purpose. At the center point, the (non-inverted) channel output is low when the (N+1)th bit is 0 and high when the (N+1)th bit is 1, except
for cases of 0% and 100% duty cycle. Prior to inversion, an up edge sets the channel to logic high, and a down edge clears the channel
to logic low.
Down edges occur when the (N+1)th bit in the PCA0 counter is one and a logical inversion of the value in the module’s PCA0CPn register
matches the main PCA0 counter register for the lowest N bits. For example, with 10-bit PWM, the down edge occurs when the one’s
complement of bits 9-0 of the PCA0CPn register match bits 9-0 of the PCA0 counter and bit 10 of the PCA0 counter is 1.
Up edges occur when the (N+1)th bit in the PCA0 counter is zero and the lowest N bits of the module’s PCA0CPn register match the
value of (PCA0 - 1). For example, with 10-bit PWM, the up edge occurs when bits 9-0 of the PCA0CPn register are one less than bits
9-0 of the PCA0 counter and bit 10 of the PCA0 counter is 0.
An example of the PWM timing in center-aligned mode for two channels is shown here. In this example, the CEX0POL and CEX1POL
bits are cleared to 0.

F850 Center Aligned PWM
When configured for center-aligned mode, a module will generate an edge transition at two points for every 2(N+1)
PCA clock cycles, where N is the selected PWM resolution in bits. In center-aligned mode, these two edges are
referred to as the “up” and “down” edges. The polarity at the output pin is selectable, and can be inverted by setting
the appropriate channel bit to ‘1’ in the PCA0POL register.
The generated waveforms are centered about the points where the lower N bits of the PCA0 counter are zero. The
(N+1)th bit in the PCA0 counter acts as a selection between up and down edges. In 16-bit mode, a special 17th bit
is implemented internally for this purpose. At the center point, the (non-inverted) channel output will be low when
the (N+1)th bit is ‘0’ and high when the (N+1)th bit is ‘1’, except for cases of 0% and 100% duty cycle. Prior to
inversion, an up edge sets the channel to logic high, and a down edge clears the channel to logic low.
Down edges occur when the (N+1)th bit in the PCA0 counter is one, and a logical inversion of the value in the
module’s PCA0CPn register matches the main PCA0 counter register for the lowest N bits. For example, with 10-
bit PWM, the down edge will occur when the one’s complement of bits 9-0 of the PCA0CPn register match bits 9-0
of the PCA0 counter, and bit 10 of the PCA0 counter is ‘1’.
Up edges occur when the (N+1)th bit in the PCA0 counter is zero, and the lowest N bits of the module’s PCA0CPn
register match the value of (PCA0 - 1). For example, with 10-bit PWM, the up edge will occur when bits 9-0 of the
PCA0CPn register are one less than bits 9-0 of the PCA0 counter, and bit 10 of the PCA0 counter is ‘0’.
An example of the PWM timing in center-aligned mode for two channels is shown in Figure 20.7. In this example,
the CEX0POL and CEX1POL bits are cleared to 0.

sskaug, 2016-05-31:
It's not specified differently, but still behaves deifferently. I think f850 uses "=" comparators (xor?) to trigger an edge, while the BBs use true "<=>" comparators. But again - AFAIK the current code should handle both. But of course, as always, proper verification is required. Probably the most stressful is multishot signaling at a high rate.

JerryXing, 2016-05-31:
So you need to validate the ESC?
This I can offer.

sskaug, 2016-05-31:
We need to validate it :)

closed 2016-07-11 by JerryXing