bitdump/BLHeli #159: Experiencing Desync on 14.6

zschiffelbein, 2016-06-03:
I am experiencing a desync issue with 2 of my quads. Both are running DYS ESCs. I can get either one of them to do it on command by perfoming a quick roll. Both quads are on betaflight 2.7.1 and BLHeli 14.6 . I have also tried BLHeli 14.1 and 14.4 and experience the same issue. Rather than downgrading both of my quads to betaflight 2.6.1 and masking the problem I would prefer to get the desync resolved. Please let me know your thoughts.

Build 1 - ImpulseRC Alien 6"
* SPRacing F3
* DYS XM30A
* EMAX RS2205 2300KV
* DAL 6045

Build 2 - Lumenier QAV210
* Naze32 Rev6b
* DYS SN20A
* EMAX RS2205 2300KV
* DAL 5045

BLHeli Settings
Demag: Low
Motor Timing: Medium-High

Here is a [Video](https://www.youtube.com/watch?v=AAiK9NVW2ZE) of the Issue. Here is a [Blackbox](https://www.dropbox.com/s/k2r02on393geyz8/blackbox_log_2016-05-31_120034.TXT?dl=0) of the issue.

mikeller, 2016-06-03:
I think betaflight improved desync protection in 2.8.0. Maybe try that?

zschiffelbein, 2016-06-03:
Thank you for the reply @mikeller . I'm fairly certain that 2.8.0 will help the issue. From what I understand betaflight 2.6.1 was sensing for a desync and lowering the power to mitigate it. 2.8.0 is slowing down the rate of change to the ESC to help prevent a desync. Pardon my ignorance, but I guess I am trying to figure out if there is a bigger issue here or is it simply the job of the flight controller to make sure that a desync does not occur?

mikeller, 2016-06-03:
Ah I see. I have pondered the same question in the past. I think the usefulness of using the extended version of a protocol (PWM) that was chosen while ESCs / servos were still analogue devices is slowly coming to an end. Using bidirectional communication between FC and ESCs would probably allow for improved control of the power output.

larsnordstrom, 2016-06-16:
Most desync is on motor 2 or 3 from what I have experienced and read so my theory is that it is caused by reversing the motor direction in BLHeli software rather solder motor wires. I had desync issues and after re-solder my motor wires the desync was gone.

This is my setup:

ZMR-X210
X-racer F303 @4k
Dys xms30a
Emax rs2205 2300kv
Betaflight 2.8.1
BLheli 14.6 Dafault settings.

4k, 2016-06-16:
@larsnordstrom hello, what can I do for you today?

zschiffelbein, 2016-06-16:
@4k , I believe you were tagged on accident. 4K is in reference to a flight controller looptime.

zschiffelbein, 2016-06-16:
@larsnordstrom , that is a very interesting thought. I too reverse motors 2 and 3 in BLHeli rather than crossing the wires. Next time I hear of someone having an issue I will recommend this to them.

le3dmax, 2016-06-16:
Interesting. I'll do that in one of my quads to see if I can solve the
desync problem. Thanks.

trojanGoat, 2016-06-16:
Seems like xm30a doesn't play well with anything above 14.4  I had Blheli 16 with betaflight 2.8.1 did all sorts of testing and nothing worked.  All problems went away as soon as I flashed 14.4 on the ESC.  I was about to start disassembling, good thing I tried this first.
Before:
https://www.youtube.com/watch?v=FIvmAZ19oRE

After:
https://www.youtube.com/watch?v=nbGbhat4-DM

sskaug, 2016-06-17:
This may be a symptom of margins. Slightly better in som cases when reversed, slightly better in some cases on 14.4. But in my opinion, there is a fundamental flaw in the XMS30A:
http://www.rcgroups.com/forums/showpost.php?p=35036613&postcount=2532

DeftWolf, 2016-06-17:
Got the same issue, sever desync issue.
I have tried to change timing and adjust PID without any improvement.
My config is:
Lux
DYS XSM 30A
DYS SE2205
Reverse 2 and 3 with Blheli
I will try tomorrow to reverse wires.

DUNcANPELTENBUrG, 2016-06-19:
I was about to post videos and BB logs, but after reading this I'll try reversing motorwires first. This "sync" issue is driving me up the freaking wall.
Martian II
DYS XMS30A
EMax RS2205 2300kv
DAL T5045BN (v1 and v2)
X-Racer F303 (BetaFlight v2.8.1)

DUNcANPELTENBUrG, 2016-06-19:
Same here! **Problem is now solved** by changing motor direction to normal in BLHeli on all 4 DYS XMS30A's and changing motor wires to change motor direction.

DeftWolf, 2016-06-20:
I have tried reversing wires but the issue remained.
One of the motors os getting very hot
I ordered a spare to replace, might be a defective one

DUNcANPELTENBUrG, 2016-06-24:
@DeftWolf Are you sure it's not a Lux (gyro) issue?

DeftWolf, 2016-06-25:
Dun, thank you for your reply. Problem is now solved for me.
After replacing the bad motor, there was no improvement (except that it was not getting hot anymore), I went back to clean flight and tried rewrit as it'much less sensitive to bad pid. The quad was flyable with a pitch and roll P of 8, shaking a lot.
I then changed the routing of one wire connected to the lux to reduce vibration. It improved a bit but not that much. 
I calibrated the esc through clean flight after removing one shot and been able to lower the P.
I moved to beta flight 2.6.1 luxfloat. Changed the esc timing to medium. Almost flyable after tuning the pid.
Then I went back to 2.8.1 in 8kHz mod to get rid off noise. The quad was jittering crazy in the yaw. I remembered Boris introduced d term in the yaw. I lowered the P and D in yaw. It flys butter smooth now!

therealjoshuad, 2016-06-26:
I recently upgraded to 2.8.0 and BLHeli 14.6 and am having the exact same issue as OP, I never used the reverse motor direction on the ESCs, I broke 2 frames trying to figure it out however, so I won't be able to test anything for a while

kidBrazil, 2016-08-27:
Hello, I also have an issue similar to OP.

I am having a very strange issue with my quad and I can't for the life of me figure it out. I came from cleanflight and this issue didn't start until I flashed betaflight.

I love the way the quad flies on betaflight its great except for the fact that when I do a full stick left-roll (very rarely happens going the other way) the quad loses control and spins on that roll direction uncontrollably. I am usually able to save it because I am not trying this low to the ground until i can fix it.

After exploring blackbox I discovered that Motor #4 was reading FULL THROTTLE but wasn't actually spinning ( otherwise the quad would not continue rolling left..). I changed ESC's and Motors on the #4 channel and nothing.

Then I replaced the FC with a brand new BRF3 and still the problem persists.. now I am at a loss.

I will list my specs and provide CLI dump + Blackbox data and ESC setting sheet below so hopefully someone here can help me figure this out =(

Helipal SRD 195
BRF3 FC + BRF3 PDB OSD combo - Betaflight 3.0 RC11
Raptor 30Amp ESC in Oneshot125 ( flycolor ) Running BlHeli 14.7
Storm 2204/2300 Kv Updraft Cooled Motors ( Helipal )

Video of the behavior -> https://www.youtube.com/watch?v=fgh0embqzuI
Link To Blackbox File -> https://www.dropbox.com/s/v4irxxojtod1e58/blackbox_log_2016-08-26_181901.TXT?dl=0
Link To CLI_dump -> https://www.dropbox.com/s/g8rohp0bvzce1r9/CLI_DUMP.txt?dl=0
Link To ESC settings -> https://www.dropbox.com/s/ecwybkq4ju254m2/ESC.PNG?dl=0

tdr250frames, 2016-09-14:
Ive the same issues as described can't sync it the motors ....

I've the sp3 seriously pro FC and have reflashed it with beta flight 3.0.0 and also tried to migrate to Cleanflight but the same thing Happens....

Motor2 is revs up uncontrollably ....

Can only find 3 out of 4 esc in " blhelisuite 16 "

A thought is if it matters that I flash one ESC at the time and that becomes MASTER...meaning that all esc is Master after flash?

I've been using the recommended flash file and all esc with settings saved to INI file in blhelisuite ......

But it's impossible to make the motors to sync .....what shall I do can I test another blhelisuite file for some like standard esc or does that ruin my hardware?

I'm lost like a parrot in Alaska ......!

Please help me!

tdr250frames, 2016-09-14:
The motors sync but they are not near synced......

I've the tx programmable checked in blhelisuite....!