bitdump/BLHeli #165: Weird ESC/motor problem

Prowlr, 2016-06-24:
I've been experiencing this problem that I haven't been able to figure out for weeks now. It's my first build and I haven't got it to a flyable point yet. Everything works apart from its ability to actually 

Once it was built I had my maiden flight. I arm, the motor spin at idle, i increase the throttle and the motors increase in speed by themselves and it flips aggressively. So I take the props off and test how the motors behave with different stick positions.

When I arm, the motors spin at idle like they should but when I adjust the throttle alone, it spins up and down like you would expect it to with all the motors in sync, but as soon as I put a little stick input that isn't throttle (yaw, roll or pitch) the motors begin to slowly increase until some of the motors are at max throttle and the others are about mid throttle. 

In CleanFlight, the motor bars go out of whack and the motors become out of sync. It's weird because when I switch flight modes in Cleanflight from horizon to angle to rate mode, the rate at which the throttle increases by itself changes. In rate mode, it goes up much slower than in horizon mode. I had the same issue with my KISS FC and switched to the Naze32 to try and figure out what it was. 

I've flashed and calibrated the ESCs through BLHeliSuite and I've searched online and couldn't find anything. All the connections are correct. Board orientation is correct. Can't see any shorts. Replaced the ESC that was causing me problems earlier but it made no difference.

I feel like it might be the PIDs but I have no idea how to tune them  
[Here is a video that shows what I see.](https://www.youtube.com/watch?v=5D3uHKBQbG8&feature=youtu.be)

Does anyone have any ideas?

prologicon, 2016-07-01:
I don't think the problem is caused by ESC, 
you can check the signal to every ESC

Prowlr, 2016-07-01:
Turns out there wasn't really much wrong with it in the first place. I tried to actually make it hover and it flew beautifully. I did replace the one lagging ESC though and now everything's perfect.