bitdump/BLHeli #167: Emax Lightning 20a

Iprenen, 2016-07-06:
Hi! 

I have trouble to flash my Emax Lightning 20A escs, which gave me magic smoke... 

I wonder why they aren't added in the latest release, I can't find the hex file in the list and the suite identifies the escs as Nano... 

Also, I have tried to add the lightning hex from the repository but I get "Read Hex File Error. Operation Aborted." 

What to do? 

P.S. I downloaded the latest from the website 5 minutes ago.

Iprenen, 2016-07-07:
Okay, so, after some debugging and recompiling, I've found the problem. 
Blheli version 16.1.14.6.0.0 is faulty. The Emax Lightning Hex have not been compiled into the BlHeliSilabsHex.dll so it instead flashes Nano20A to the ESCs. The ESCs can handle them but it will fry your motors. Recompiling with the correct hex or flashing with the hex files in the repository (dumping them into the BlHeli folder and choosing Flash Other, you can flash the correct hexfile to the ESCs. 
Four new RS1306 motors coming up...

4712, 2016-07-07:
BLHeliSuite version 16.1.14.6.0.0 is not faulty.

For the suggestion of a firmware to update, BLHeliSuite needs reliable data. So, if it finds an ESC flashed with XYZ firmware version 14.5, it will suggest XYZ firmware version 14.6. If it finds a bootloader compiled for a certain mcu, it will only show a selction list of firmware suitable for that specific mcu. 

If the ESC was delivered with the wrong firmware (which is for the Emax Nano), BLheliSuite will recognize it as a F330 type ESC and will not show up the F396 (Emax Lightning) firmware in the list.

Where did you get the EMAX Lightning 20A from? AFAIK it has a F396 mcu whereas the EMAX Nano is a F330 mcu ESC.

Iprenen, 2016-07-12:
I thought it was strange too, that they id'd them as Nano as they are not the same at all. 

The main problem was that EMAX Lightning hex wasn't in blheli suite 16.1.14.6.0.0 at all. I checked all the hex files in the precompiled version, all the ESCs was in there except for just the Lightning ones. That's must be why it couldn't recognize them. As I added the hex files from another source, they flashed and updated beautiful. But still running Lightning firmware, if I use the Flash Blheli instead of Flash Other, it chooses the Emax Nano firmware. That's why I had my conclusion that the precompiled version was faulty and I have not had the time to test the 16.2.14.6.0.0 yet.

4712, 2016-07-12:
The only chance to make it an ordinary EMAX Lightning 20A, is to flash it with a SiLabs C2 capable interface, so that the bootloader is also exchanged against the one compiled for the SiLabs F39x chip.

Otherwise, BLHeliSuite 16.2.14.6.0.0 now has the opportunity to flash hexfiles compiled for F39x to an ESC which identifies itself as a F33x based one.
Enable "Expert mode" to enable the "Ignore inappropriate MCU" check box...

[image]

HomeY, 2016-07-21:
Also had the magic smoke yesterday when i upgraded my ESC's. Mine had 14.2 FW on them and also got recognized as Nano. And just now i ran into this message from EMAX:
http://www.emaxmodel.com/blog/AmandaChen/

That might explain why my RS2205 got incredibly hot after just 5 seconds of 3s lipo power (without spinup).
Let's hope my motors are fried!

4712, 2016-07-21:
Yes, errors happen, but this one was really needless. Never again please.

SRTDRE76, 2017-03-26:
Watch drone channel and YouTube segments before trying any upgrading. There's a vid for whatever you got and where you want to take. I New as well but did my watching researching and reading so not to go gun ho and throw $$ away