bitdump/BLHeli #168: Changing motor direction without using blheli suite?

vcazan, 2016-07-09:
I am developing a hardware prototype where I need to be able to change the motor direction config using a push button and an arduino? Can anyone point me to where I can get some more information on where the config values are stored on the EEPROM? Thank you very much.

vcazan, 2016-07-09:
Never mind i've found it here: https://github.com/bitdump/BLHeli/blob/master/Atmel/BLHeli.asm#L1268

jackbnimbl, 2016-07-10:
These needs to be a thing like right now. And betaflight needs to use it. Use a switch to reverse motor direction and flip over downed quads during a race. It's silly that a quad that lands up side down can't right itself.

tracernz, 2016-07-10:
@jackbnimbl You are aware of 3D mode?

jackbnimbl, 2016-07-10:
Seems a little much to switch into 3D mode (can it be attached to a profile/switch?) just to right a crash no? plus I would need compatible escs and practice learning 3D mode.

tracernz, 2016-07-10:
You could have your flight firmware only use the positive thrust range normally, until it needs to right a flipped quad, then it can use reverse. ;-)

prologicon, 2016-07-25:
1.change the value of "Pgm_Direction"
2.decode the direction
3.save the direction number to eeprom.

KnuckleUpFPV, 2017-08-15:
There is dshot cmd to flip a switch and reverse motor direction now. Have to flash with 16.67 unless on blheli_32 then you need the beta firmware and the wraith 35a. In betaflight 3.2 modes tab add dshot motor reverse to a switch. Disarm flip switch rearm throttle up to flip over. Once upright disarm flip switch off rearm fly away