bitdump/BLHeli #171: Clarification in BLHeli_S manual SiLabs documentation

robomaniac, 2016-08-15:
Hi

This section of the documentation need some clarification or an example
[BLHeli/BLHeli_S SiLabs/BLHeli_S manual SiLabs Rev16.x.pdf](https://github.com/bitdump/BLHeli/blob/master/BLHeli_S%20SiLabs/BLHeli_S%20manual%20SiLabs%20Rev16.x.pdf)

> Code naming convention:
> The BLHeli_S codes are named with a letter, another letter and two numbers, in
> addition to the revision. For example "A_L_10_REV16_0.HEX".
>  **The first letter denotes the pinout of the MCU.**
> The second letter is either L or H. L is for 24MHz MCUs, and H is for 48MHz
> MCUs.
> The two numbers denote the FET switching deadtime. The unit is 20.4ns.
> Some FET drivers have adaptive FET deadtime control, and for these 00 is used
> to denote FET switching deadtime. 

"_The first letter denotes the pinout of the MCU_" 
For me that is not clear. I have an Aikon 30A and the MCU is an [EFM8BB2] What is the pinout?(https://www.silabs.com/Support%20Documents/TechnicalDocs/EFM8BB2-DataSheet.pdf).

**Could we make this more clear by giving a better description or giving an example?**

also about the switching deadtime, how can someone calculates or know which is the proper value for his ESC?

JerryXing, 2016-08-15:
Deadtime is the hardware design decisions.

tracernz, 2016-08-15:
Deadtime is the time between switching off one side of the FET bridge, and switching the other side on, because FETs take time to "switch on" and "switch off" (pesky physics). Too short and you get shoot-through, current that "shoots" through both FETs, i.e. a short-circuit. Depends on hardware design and components.

closed 2016-10-02 by robomaniac

DroneMesh, 2019-04-19:
No one answered the pinout.. I need this as I am releasing open hardware blheli_s esc schematics in next couple days

robomaniac, 2019-04-26:
hey @DroneMesh, the pin out are all in the .asm (assembly) code
take any random version og blheli for example
https://github.com/bitdump/BLHeli/blob/master/BLHeli_S%20SiLabs/G.inc

line 131 is port 0 pin 5 that is the RC input
line 321 is port 1 with all your mosfet pinout, the PWM and common pins for each mosfet phase.

you will need to flash the .G version of blheli. Because open other letter and it's different pinout.

robomaniac, 2019-04-26:
I just saw your video, you figure it out. cool

DroneMesh, 2019-04-26:
Hi,

Yea took couple hours but I ended up figuring out that it's by the pin port
if the datasheet.