bitdump/BLHeli #172: OCDAY 20A ESC's wich firmware BLheli

Koekebeest, 2016-08-16:
Hello, some time ago I ordered these ESC's http://www.gearbest.com/multi-rotor-parts/pp_359436.html and I was upgrading the firmware in BLHelisuite but it's says it's running on Afro 12A firmware. Can I upgrade these ESC's to Afro 20A firmware? Since the esc's are 20A but then OCDAY 20A. Or is there an Ocday compitable firmware? 

Does anyone have an idea? Sure hope so, I wan't dampening running in a save way. Without frying anything.

I hope you can help me out.

Greets!

nlbaviu1, 2017-01-05:
Did you already figure this out? I am struggling with the same question....

Koekebeest, 2017-01-06:
I believe someone has tried and upgraded it to 20amp and succeeded. I'm not a big fan of trying those things out so I didn't do it. It flies great with dampening on. So I suggest just leave it at 12A afro and do dampening on.