bitdump/BLHeli #175: Flash verify failed error  -  on blheli_s 16.3

blazomusic, 2016-08-25:
Guys, I have a serious problem. 

16.3 came out, and I flashed my racerstar 30A blheli_s v2 ESC's with G-H-30 firmware.
I flashed ok, then I disconected, connected again, and 3 of 4 ESCs are in blheli bootloader, don't have the correct firmware. 
What is worse, I cannot flash them again because there is an error : 
"blheli, failed to write flash, Flash verify failed error "

I tried to flash with 16.2 and there is the same error. 

I connected to blheli one more time, now all 4 of the ESCs are in the bootloader, and I can't flash. 
Please help me. I am trying with fc (betaflight 2.9.1) passthrough.

Please help me with this, is this a 16.3 bug or what?

Orrrb, 2016-09-01:
I have the same issue using passthrough, usb programmer and an arduino nano on fvt 30a escs. They came with 16.2. No way this will flash. Had it working on 16.3, needed to reverse two motors, connected in passthroug and bam "wat is dis esc? No config for you!"

illest33, 2016-09-02:
The latest BLHeli has a bug that erases the flash unless you unplug the lipo before disconnecting from BLHeliSuite. I have bricked 4 xs20a because of this. You now need to use a arduino with c2 pin connected to bring it back.

Orrrb, 2016-09-02:
Do you have a source please, illest33? There are very few resources about this, at least with my googlefu, and those that exist all refer to the c2 method or are just reports without replies. 

I have tried both the arduino c2, the fvt usb programming dongle with blheli and FVT`s own software and passthroug w betaflight, to no avail. BLHeliSuite reports "blheli bootloader" but unknown firmware.  It doesnt seem to have erased the firmware though, as I am flying them, but I cant connect to change anything ofcourse.

blazomusic, 2016-09-02:
Hey. I had the same problem but I managed to flash them using pads on esc
with c2 interface with arduino nano. No problem. I dont have any other
ideas

So we should disconnect battery first before clicking disconnect in blheli
in the future?

illest33, 2016-09-03:
Orrb I saw your post this morning, if you had 16.0, 16.1, or 16.2 you should be able to right click the "unknown firmware" and flash it in single. Make sure you select the right letter that corresponds to the pinout and clock speed of the mcu then number. If you are on 16.3 and receive the "erased or unknown firmware" message you have to recover with a arduino on c2 pin somehow. I haven't even tried recovering my xs20's yet. It's a problem with certain F3 boards and the pass through along with blheli. So just remember when you get the firmware loaded back, always disconnect the battery before you click the "disconnect radio button" and close blheli suite. I would NOT recommend flashing to 16.3.

4712, 2016-09-05:
Issue is fixed with BLHeliSuite 16.3.14.7.0.2

Orrrb, 2016-09-05:
Very nice! Just tried it out, worked flawlessly this time, running 16.3. Thank you for your effort 4712.

aalec25, 2016-10-06:
BLAZOMUSIC

Hello,

I have the problem as you. I bought a Arduino Nano too.

Was is C2 Interface ?

Thanks you in advance,

AleC

Orrrb, 2016-10-07:
I got it fixed with a nano and the C2 interface. ESC`s have been going strong ever since. Take a look [here ](https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeliSuite%204w-if%20interfaces%20pinout.pdf) and [here](https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeliSuite%204w-if%20interfaces%20pinout.pdf)

aalec25, 2016-10-07:
Hello Orrrb, 

Thanks your for you help.

My ESC DYS XM 10A are supported by blheli.

If I have well understood about Arduino Nano ATmega328P , I need to weld the black on Ground, and the White on one pin from D4 to D10 because of multi. 
And in Blheli, I need to use SILABS 4way-if NOT SILABS USB/COM. 

AleC

aalec25, 2016-10-11:
Hello,

Unfortunately, it changed nothing.

What can i do ?

thanks,

AleC

4712, 2016-10-11:
@aalec25 this might help https://www.youtube.com/watch?v=b-7VFt3IhhE

aalec25, 2016-10-12:
@4712 

Thanks for your link ! I have tried but no success.

I send this message to the author of the video. I'm waiting for his answer & let you know.

"Hello ! 

Thanks you for the great video ! 

I tried your tutorial for a ESC DYS XM10A. There are the 4 pads. I used the second one from the left to weld the C2C cable, the third one for the 3,3 V from Arduino Nano & the last one for the ground. 

The arduino NANO 328 had been flashed with Arduino 4way-interface and 4wArduino_Nano__16_MULTIv16400. 

I have connected the Esc to flash it, Blheli recognized the arduino but when I want to read it tells : 

"Serial Interface"m4wANm328P16" connected succesfully. 

Connexion to ESC#1 failed! 
Please check data- und power-connections to the ESC! 

Hint : All interface lines are LOW: 
ESC seems to be disconnected or not powered up" 

I have the same when I put out 3.3 V and use the lipo. 

Do you have any idea about the problem ? 

Thanks you in advance, 

AleC﻿"

4712, 2016-10-12:
@aalec25 Do NOT connect 3.3V. The read marked pad does NOT mean VCC!
Please read here https://nathan.vertile.com/blog/2015/11/16/flashing-xrotor-20a-with-blheli-via-4w-if/#flashing for better understanding.

aalec25, 2016-10-12:
[image]

Hello,

My XM10A has only 2 cables : 1 black which must be Ground and 1 yellow which must be use as C2CK.

The arduino is well programmed. Just need to know where I must weld these 2 cables.

I will use my lipo as power.

Thanks,

AleC

4712, 2016-10-12:
Your link does not work.
The cables cannot be used for the C2 flashing. You need to connect either over the bootloader using the interface "**D** SiLabs BLHeli Bootloader (4way-if)", or remove the heat shrink and solder direct to the C2 interface pads on the ESC board. I assume, you would need further help for this...

aalec25, 2016-10-12:
@4712 

The link doesn't work ? https://we.tl/2qpkXmfkKs

I would like to thank you for your time & help,

AleC

4712, 2016-10-12:
Ok now I can see. The gnd pad is right (black). The order is GND nc C2CK C2D.

aalec25, 2016-10-12:
@4712 

Thanks,

So I have well understood, from left to right, it is C2C - C2CK - Nc - GND.

I need to connect C2CK + GND, and flash.

Sorry my noobness, but it is the last point to understand before to be "independant"...

4712, 2016-10-13:
Did you read the tutorial (link above)? 
You need C2D, C2CK and GND.

aalec25, 2016-10-13:
@4712 

Thanks you very much for your patience !!! I did it successfully !!! My mind was blocked because my ESC has only two cables, but why don't weld a third one more ? Thanks you for your perseverance !

I resume shortly  for people who has the same problem :
- Buy an arduino Nano ATmega328P V3
=> it uses a ch340g usb chip, the driver for which is here: http://www.wch.cn/downfile/65

- Download the lastest version of blheli here :  blhelisuite.wordpress.com

- Use the program to flash your arduino Nano in 
1) Connect the arduino & select the good communication port
2) Select Make Interfaces
3)  Select Arduino 4way-interface
4) Ok
5) Select 4wArduino_Nano__16_MULTIv16400.hex & confirm

- You will find 4 pads on the board of your ESC. For mine, from left to right, you have to connect/weld C2C - C2CK - Nc - GND
- You can use this link [https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeliSuite%204w-if%20interfaces%20pinout.pdf](url) to help you to know where to connect the cables on the arduino.
- To be sure if your ESC is supported,please check here : [https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeli%20supported%20SiLabs%20ESCs.pdf](url)
- When the THREE cables are connected to the arduino Nano, connect you on Belheli, connect, read setup, and choose the good firmware in Multi for your ESC, confirm & it must be well flashed
=> Link from 4712 = Please read here https://nathan.vertile.com/blog/2015/11/16/flashing-xrotor-20a-with-blheli-via-4w-if/#wiring for better understanding

Please free to contact me if any trouble, and thank the community for the help !!

See u,

AleC

Tadinius, 2017-02-05:
Hi all, I know it's an old post, but I have same or similar issue with blheli s 4in1 10a esc, tried to flash it in blheli, esc #2 failed flashing and now its not recognizable in suite, I have an arduino and thought if I could get it flashed and working again.. I will get it apart to see if there are any pads for connecting to arduino as the esc so tiny.. maybe u guys could help me to recognize the pins layout on esc? Not sure where to cennect it. Thanks

aalec25, 2017-02-07:
Hello,

On my Esc, on pads side when the cable are on the left, the order of the pads is GND nc C2CK C2D.

Hope it will help you,

AleC

Tadinius, 2017-02-10:
hi again guys, as mentioned before, I've got blheli 4in1 10a esc, which one esc failed to flash, now trying to flash it using instructions stated in this post and still cant get it recognized by blheli, as you see in the picture there are two pads near each esc, these must be c2ck and csd pads, gnd goes to the port, ive tried to connect it using 4 way interference and also other ways but esc just cant get recognized, maybe someone got any ideas how to fix it as its driving me crazy lol.. thanks
[image]
[image]

FlyerFpv, 2017-06-20:
Hi, 
Bhelisuite bricked 3 of 4 XS30A escs, now I cant flash it.
I will try C2 mode.
Anyone know the if the sequence of the signals pads in the picture are right? 
Regards
[image]

illest33, 2017-06-21:
yes, those are the pads to use. Just look up the pinout on google.

[19238384_1746349435393559_450202073_o]<https://user-images.githubusercontent.com/20193680/27356333-fdbacaba-5605-11e7-9362-0914e21366e8.jpg>

voodoo614, 2017-06-22:
You can determine C2 port by testing for the ground. Sequence goes Ground, 3.3V, C2D and C2CK.

FlyerFpv, 2017-06-22:
voodoo614, what do you mean with testing for the ground?

voodoo614, 2017-06-22:
Using a multimeter, determine which pad is the ground. The rest follows in the order I wrote down above.

FlyerFpv, 2017-06-22:
Ok ok, Understand :+1: Thanks
Regards

FlyerFpv, 2017-06-23:
Hi, I ask the dys team and Im confused with it.
which one is correct? :)
[image]

4712, 2017-06-23:
HTH
[image]

FlyerFpv, 2017-06-23:
How can I see the pin number 1? to much small on esc :(

FlyerFpv, 2017-06-23:
Thanks 4712.
I found the pin number 1.

4712, 2017-06-23:
The orientation is the same. Top is GND.
You do not use VDD.
Use
GND = black

C2CK = red
C2D  = white

FlyerFpv, 2017-06-23:
Thanks for your time.
Best regards

FlyerFpv, 2017-06-24:
Successful flash. Thanks guys for your help.
[image]
[image]