bitdump/BLHeli #178: Rapid throttle reset

KyunghyunLee, 2016-08-31:
Hi, 
I am making DIY ESC with BLHeli_S.
Motor beeps when I connect battery, and turn correctly according to throttle value.
However, ESC suddenly reset(beep start again) when sudden throttle raise.

I made DIY ESC with BLHeli successfully before, but I am in trouble with BLHeli_S.
And also, I am not familiar with assembly code, I can just read a little and guess what the code is. 

What is condition for sudden reset in BLHeli_S? 
Or, is there any same reported problem with other ESCs?

sskaug, 2016-08-31:
Since your ESC beeps, this indicates that the MCU resets. Which is due to the MCU supply voltage dipping, causing a brownout reset of the MCU.

KyunghyunLee, 2016-08-31:
Thank you for your reply.
I found some other ESCs have same issue, and they solve it by changing regulator. 
I also changed my regulator, my problem is gone.

Thank you!

closed 2016-08-31 by KyunghyunLee