bitdump/BLHeli #180: Flash verify failed error - on bLheli 16.3.14.70.2

aalec25, 2016-09-23:
Hello !!

I successed to flash DYS ESC XM10A (presents in the manual) with the last version 14.7, of x Rotor 10A Multi but the next day, for a last check, two Escs 2 & 3 weren't respond, so I decided to flash them again.

Since, the program doesn't recognize any firware and for any new try of flashing, I have " Flash verify failed" error.

I don't really know why can i do.

If anyone can give me some tips, thanks you in advance,

Have a good day,

Alex

4712, 2016-11-02:
@aalec25 Always use the latest [BLHeliSuite ](https://www.mediafire.com/folder/dx6kfaasyo24l/BLHeliSuite)please. BLHeli 16.3./14.7 SiLabs contains a bug, which might lead to bootloader corruption. 
If bootloader is corrupted, You will need to reflash the ESC using a C2 connection.

aalec25, 2016-11-03:
@4712 

Thanks for the tip again.

Because of his bug, I lost 3 ESCs... I can't reflashed them again, in C2 connection too..

4712, 2016-11-04:
@aalec25 C2  flashing almost always works...