bitdump/BLHeli #181: BLHeli support for SPEDIX Round 12A ESCs

Ciolouse, 2016-10-10:
Greetings,

Can you add the Round 12A ESC from SPEDIX to your database please ?

http://www.hobbyking.com/hobbyking/store/__82586__Round_12A_ESC_with_Green_LED_For_Spedix_Series_Multirotors_1pc_.html

http://www.hobbyking.com/hobbyking/store/__82585__Round_12A_ESC_with_Red_LED_For_Spedix_Series_Multirotors_1pc_.html

Thanks a lot !!

Ciolouse, 2016-10-11:
This ESC is the same as this one :

http://www.buddyrc.com/velotech-magic-multirotor-speed-controller-12a-with-1a-bec-simonk-program-and-bullet-connectors.html

Ciolouse, 2016-10-31:
We can use AFRO_12A config but the led are not working anymore ...

Ciolouse, 2016-11-02:
I know it's more a request than a issue, but can you have a look of it please ?
Thanks a lot !

ariapro22, 2022-08-11:
Did anyone manage to flash these esc’s. I’ve just built a spedix hexacopter kit with these and luminer f7 pro inav but they arrived with older SimonK