bitdump/BLHeli #182: Magic smoke after setting beep strength to 80 on 14.8

openscript, 2016-10-15:
Dear community

I've been flying with my EMAX Lightning 30A ESCs for a while now. Today I updated them from BLHeli 14.2 to 14.8. I've also activated damped light and set the beep strength to 80. 

When I powered them up after the update some magic white smoke came out of the battery. The motors got really hot. Of course I disconnected the battery immediately. 

Then I unsoldered all motors from the escs. Flashed 14.2 again and soldered one motor to an esc. When I powered that on again, the ESC started to smoke.

I've found this: http://www.emaxmodel.com/blog/AmandaChen/

So I flashed the wrong firmware on the ESCs?

Robin

Madforit966, 2016-10-17:
I have the same problem nearly £50 down the drain Have looked at the boards and found small beads of solder Now touch across 2 legs on one of thge chips Causing a short This has happened on all 4 of mine I have remved with some effort this solder I think it's caused by it overheating Then melts some solder off the board and this is were it ends up 
I just need to re flash Trying through arduino nano but gettin g failure
Help!

openscript, 2016-10-20:
I could save 2 of the ESCs. I've flashed https://github.com/bitdump/BLHeli/blob/master/SiLabs/Hex%20files/EMAX_Lightning_20A_MULTI_REV14_8.HEX and they started to work again. The other two have a hardware failure.

openscript, 2016-10-20:
Due the high current the PDB was roasted as well.