bitdump/BLHeli #183: codes under label "calc_next_comm_startup"

nonamerubbish, 2016-10-16:
It seems that Temp2:Temp1 was limited to 0xFFFF in some case from line 3993 to line 4001. However, it is then changed to Prev_Comm_H:Prev_Comm_L. Is there something, or what is the purpose?

sskaug, 2016-10-17:
Good point. Should rather be:
_calc_next_comm_startup:
	mov	Temp6, Prev_Comm_X
	mov	Prev_Comm_X, Temp3		; Store extended timestamp as previous commutation
	mov	Temp2, A
	mov	A, Temp3
	subb	A, Temp6				; Calculate the new extended commutation time
IF MCU_48MHZ == 1
	anl	A, #7Fh
ENDIF
	mov	Temp3, A
	jz	calc_next_comm_startup_no_X

	mov	Temp1, #0FFh
	mov	Temp2, #0FFh
        ajmp	calc_next_comm_startup_average

calc_next_comm_startup_no_X:
	mov	Temp7, Prev_Prev_Comm_L
	mov	Temp8, Prev_Prev_Comm_H
	mov	Prev_Prev_Comm_L, Temp4
	mov	Prev_Prev_Comm_H, Temp5
	mov	Temp1, Prev_Comm_L		; Reload this commutation time	
	mov	Temp2, Prev_Comm_H
	clr	C
	mov	A, Temp1
	subb	A, Temp7				; Calculate the new commutation time based upon the two last commutations (to reduce sensitivity to offset)
	mov	Temp1, A
	mov	A, Temp2
	subb	A, Temp8
	mov	Temp2, A

calc_next_comm_startup_average:
	clr	C
	mov	A, Comm_Period4x_H		; Average with previous and save
	rrc	A
	mov	Temp4, A
	mov	A, Comm_Period4x_L
	rrc	A
	mov	Temp3, A
	mov	A, Temp1			
	add	A, Temp3
	mov	Comm_Period4x_L, A
	mov	A, Temp2
	addc	A, Temp4
	mov	Comm_Period4x_H, A
	jnc	($+8)_
When there is a difference in the Xtended byte, the commutation time is set to max before averaging

nonamerubbish, 2016-10-18:
Then, It changes like this?**(Atmel/BLHeli.asm)**

3993 calc_next_comm_startup: 
3994 	lds	Temp6, Prev_Comm_X 
3995 	sts	Prev_Comm_X, Temp3		; Store extended timestamp as previous commutation 
3996 	sbc	Temp3, Temp6			; Calculate the new extended commutation time 
3997 	tst	Temp3 
3998        breq calc_next_comm_startup_no_X (***)
3999 
 
4000 	ldi	Temp1, 0xFF 
4001 	ldi	Temp2, 0xFF 
+++         rjmp calc_next_comm_startup_average
4002 
+++ calc_next_comm_startup_no_X:
4003 	lds	Temp7, Prev_Prev_Comm_L  
4004 	lds	Temp8, Prev_Prev_Comm_H
4005 ----
4006 ----
4007 	lds	Temp1, Prev_Comm_L		; Reload this commutation time 
4008 	lds	Temp2, Prev_Comm_H 
4009 	sub	Temp1, Temp7 			; Calculate the new commutation time based upon the two last commutations (to reduce sensitivity to offset) 
4010 	sbc	Temp2, Temp8  
+++ calc_next_comm_startup_average:
+++         sts	Prev_Prev_Comm_L, Temp4 
+++         sts	Prev_Prev_Comm_H, Temp5  
4011 	lds	Temp3, Comm_Period4x_L	; Average with previous and save 
4012 	lds	Temp4, Comm_Period4x_H