bitdump/BLHeli #188: [RBBX] Cant Connect to ESC

trashmaster9000, 2016-11-02:
hi,

i set up the rbbx box to change settings on my ESCs.
with my Atmel ESCs it works fine, but with my Silabs ESC (Turnigy 120A-HV-OPTO v2) i cant connect to the ESC.
i tried all 3 Firmwares for the RBBX.

with a arduino nano and the 4Way-Interface FW (D Silabs BLH Bootloader) it works fine.

4712, 2016-11-02:
@trashmaster9000  Which version of BLHeliSuite did you use to Flash the RBBX?

trashmaster9000, 2016-11-02:
i use the latest version (BLHeliSuite16414801)

4712, 2016-11-02:
How do you connect the ESC with the box?
Servo direct?

trashmaster9000, 2016-11-02:
yes, i connect the Servo plug (the short one for programming) direct to the box.
i also connect 5v to gnd and 5v on the servo connector.

4712, 2016-11-02:
If you use the servo port then only servo direct hex would work.
But servo direct might fail with this specific ESC.
So I would suggest to modify the box input scenario (increase R8 eg.) or to use another MCU pin (PINC5) as used for C2 interface connection.

trashmaster9000, 2016-11-02:
i replaced R8 1kOhm with 10kOhm
it works now.

thanks for the fast support.

closed 2016-11-02 by trashmaster9000