bitdump/BLHeli #190: Emax Lightning 30A F390 48MHZ not enabled

jurkust, 2016-11-09:
From https://github.com/bitdump/BLHeli/blob/master/SiLabs/EMAX_Lightning_20A.inc I see that MCU_48MHZ is not enabled, but it has F390 MCU like FVT Littlebee (https://github.com/bitdump/BLHeli/blob/master/SiLabs/FVT_Littlebee_20A_Pro.inc) which has MCU_48MHZ enabled.

I ask this, because I hope that after enabling higher frequency it will support OneShot42 (I'm using kiss fc).
I can test if it works, just send me a hex file :)

sskaug, 2016-11-09:
You are right, this ESC is not set to run at 48MHz, even if it should have been.
You have an email I can send test code to?

sskaug, 2016-11-09:
Actually I can just attach it here :).

jurkust, 2016-11-09:
ESCs works with new hex, no heating or other side effects. Will do in depth test on weekend.

yo8aiv, 2019-09-19:
hi. is it possible that these EMAX_Lightning_20A support dshot ?  
i also have the FVT_Littlebee_30A wich work fine with dshot600 using A-H-20.  
hardware seems very similar. 
since its old hardware i might experiment.
thx.