bitdump/BLHeli #191: Which one?

DubLD, 2016-11-12:
I'm trying to get in on this Dshot thing. And I don't know which of those hex files I use for my ESC's? A lil help?

varkokonyi, 2017-05-06:
Right now you only have to update it (a bit late, i know)

DubLD, 2017-05-07:
Does it matter that I don't have any arduinos?

⁣-Don D. ​

https://github.com/bitdump/BLHeli/issues/191#issuecomment-299671238

varkokonyi, 2017-05-07:
Depends. What fc, what fw?

DubLD, 2017-05-07:
And will I have to go through this process every time I try to update it? Or will I be able to do it the  regular way with BLHeli Suite after this without all the funny stuff?

⁣-Don D. ​

https://github.com/bitdump/BLHeli/issues/191#issuecomment-299671238

varkokonyi, 2017-05-07:
Well, I need to know your setup to answer. What fc do you have and what software is on it?

DubLD, 2017-05-07:
Basic F3-F4 w/pass-though running betaflight 3.1.7.?

⁣-Don D. ​

https://github.com/bitdump/BLHeli/issues/191#issuecomment-299673370

varkokonyi, 2017-05-07:
Then you can just update through the suite. Choose the same edition you have already (for example, mine is G_L_30)

DubLD, 2017-05-07:
A GetFPV Basic F4 (the older white one) with BetaFlight 3.1.7. but honestly that's subject to change any time now, or it may not. We will see. But if I can use it to get the good firmware on there, then I will.

⁣-Don D. ​

https://github.com/bitdump/BLHeli/issues/191#issuecomment-299673506

DubLD, 2017-05-07:
When I try to use blheli Suite on it it tells me I have to power at Macy's even though they are powered up and it gives me some "mtlcntrl" (that's probably not exactly what it says, or how its spelled) error/warning. But if you think it'll help I can recreate the issue.

mikeller, 2017-05-07:
@DubLD: Have you tried updating your ESCs with the blheli configurator (https://chrome.google.com/webstore/detail/blheli-configurator/mejfjggmbnocnfibbibmoogocnjbcjnk?hl=en) instead?

DubLD, 2017-05-07:
Yeah, I click read set up and it stalls and does nothing.

⁣-Don D. ​

https://github.com/bitdump/BLHeli/issues/191#issuecomment-299679232

varkokonyi, 2017-05-07:
What ESCs are you trying to flash?

DubLD, 2017-05-07:
Emax Nano 12a Opto V2s

⁣-Don D. ​

https://github.com/bitdump/BLHeli/issues/191#issuecomment-299692972