bitdump/BLHeli #197: Throttle calibration broken in 16.5 when running Mutishot at a 32khz pwm rate.

cturn, 2016-12-02:
Tested 3 different flight controllers (Flip32 F4, Revolt, SPF3) and 2 different esc's (Aikon20 & Bumpbee30) and get the same results on both Betaflight and Raceflight.  Throttle calibration will not succeed when running multishot at 32khz pwm rate.  If I lower the pwm rate from 32khz to 16khz with multishot protocol selected, calibration is successful in 16.5.  

This issues is not present in Blheli_S 16.4, when I revert the esc firmware to 16.4 throttle calibration is successful on Multishot at 32khz pwm rate.

cturn, 2016-12-02:
Moving Multishot to before Dshot in the protocol detection order resolves this issue.  I'll submit a pull request shortly.

sskaug, 2016-12-04:
Generated a fix that I believe is good: https://github.com/bitdump/BLHeli/tree/master/BLHeli_S%20SiLabs/Rev16.51%20test%20code%20Multishot%2032kHz

DubLD, 2016-12-21:
Page no longer works.

skorzo, 2017-06-29:
Probably related  to https://github.com/bitdump/BLHeli/issues/213 ??

phantom-8, 2018-01-08:
Just tested the latest BLHeli_S v16.7, still has the same problem.  Will there be a fix on this?  Or do we need to stay on v16.4?  Yeah, this problem is the same as #213