bitdump/BLHeli #199: Installation error message: "Device cannot be identified"

Steadyspike, 2016-12-06:
I have downloaded Blheli 15.5.14.8 release. Once uncompressed, I can not run BLHeliSuite.exe because an error message appears: "Device cannot be identified". The details about the exception are:
  2.2 Address: 005D1536
  2.3 Module : BLHeliSuite.exe
  2.4 Type   : EControllerError
  2.5 Message: Device cannot be identified.

4712, 2016-12-06:
@Steadyspike can you try to check in the device manager, if there is any device marked as unknown or problematic?

4712, 2016-12-07:
@Steadyspike could you please post the "BLHeliSuite.**elf**" (EurekaLog) file from the BLHeliSuite folder here?

Steadyspike, 2016-12-07:
Here you have .elf (zipped) and a Domain Manager screenshoot.
[image]

[BLHeliSuite.zip](https://github.com/bitdump/BLHeli/files/635826/BLHeliSuite.zip)

4712, 2016-12-07:
I use an older Version of EurekaLog, so it shows vista max. But build number is 7601, so Win7...
Anyway, it would be helpful to exchange the BLHeliSuite.exe with the attached one and post the elf file again.
[BLHeliSuite.zip](https://github.com/bitdump/BLHeli/files/636190/BLHeliSuite.zip)

Steadyspike, 2016-12-07:
I have replaced it and doesn't work. Find attached log file.
[BLHeliSuite_v2.zip](https://github.com/bitdump/BLHeli/files/637401/BLHeliSuite_v2.zip)

4712, 2016-12-07:
@Steadyspike Thanks a lot. It is still unclear for me what happens.
There must be a HID device what cannot be opend for check.
This will raise an internal exception, normaly no problem, it should not show up...

Now, we can find the device, which causes the exception, may be this helps...

Please use the attached BLHeliSuite.exe. The exception dialog should show info about the HID device path (VID and PID). 
[BLHeliSuite.zip](https://github.com/bitdump/BLHeli/files/637965/BLHeliSuite.zip)

Steadyspike, 2016-12-08:
Replaced again but doesn't work. I hope it helps you debugging the problem...

[BLHeliSuite_log_v3.zip](https://github.com/bitdump/BLHeli/files/638058/BLHeliSuite_log_v3.zip)

4712, 2016-12-08:
Thanks. It does not bring any further information though...
Maybe I find some fix at weekend.

4712, 2016-12-10:
@Steadyspike Could you please download Device Manager View from http://www.nirsoft.net/utils/device_manager_view.html and look for any device with Device instance ID "ROOT\MEDIA\0000"?

4712, 2016-12-10:
@Steadyspike I hope this is the fix... please check.
[BLHeliSuite.zip](https://github.com/bitdump/BLHeli/files/643965/BLHeliSuite.zip)

Steadyspike, 2016-12-10:
Working !!!!!!!
Very good job!!! thanks for your effort

Which was the problem? how do you fix it?

4712, 2016-12-10:
@Steadyspike thanks a lot! It was a bug in the JVCL library (JvHidControllerClass) for Delphi...

in the constructor 

> constructor TJvHidDeviceController.Create(AOwner: TComponent;  AOnHidCtlDeviceCreateError: JvHidDeviceCreateError; AOnDeviceChange: TNotifyEvent);

the two lines

> SetOnDeviceChange(AOnDeviceChange);
FOnDeviceCreateError := AOnHidCtlDeviceCreateError;

are in the wrong order.
FOnDeviceCreateError  must be set first.
I worked around by calling the constructor in two steps:

> FHidCntrl := TJvHidDeviceController.Create(Self, DoOnDeviceCreateError, Nill);
FHidCntrl.OnDeviceChange := DoOnDeviceChange;

Steadyspike, 2016-12-10:
Final question: next BLHelisuite releases will fix it? in case it will not be fixed, could I replace Blhelisuite.exe with the file you provided me?

Thanks again!!!

4712, 2016-12-10:
Sure, next release will include the fix, but for now you can simply replace the exe.

Steadyspike, 2016-12-11:
Great!!
Thanks for your fast help

closed 2016-12-11 by Steadyspike

4712, 2016-12-11:
@Steadyspike Thanks a lot for **your** help! BLHeliSuite16.514.8.0.1 is now [released](https://www.mediafire.com/#dx6kfaasyo24l), the only change is the fix...