bitdump/BLHeli #200: BlHeli_S setting ONE_S_CAPABLE to 1 breaks bootloader

fishpepper, 2016-12-07:
Setting the ONE_S_CAPABLE flag to 1 (as used in O.inc) will break the bootloader.
Connecting and reading works fine, all flash erase or write operations will fail.
There will be no ACK/NACK after the flash command.
It seems like the erase/write operations trigger a device reboot?!

sskaug, 2016-12-08:
If I remember the manual correctly, the voltage monitor has to be enabled on the BB MCUs, in order to flash. When ONE_S_CAPABLE is set, the voltage monitor is disabled in order to avoid undesirable resets (this was done on the 1S operated mCPx).
The ONE_S_CAPABLE setting is not used, no longer very relevant, and should be removed, I guess.

fishpepper, 2016-12-08:
Thats it!

> To help prevent the accidental modification of flash by firmware, hardware restricts flash writes and erasures when the supply monitor is not active and selected as a reset source. 

I use the one_s_capable setting for my 1S esc so please do not remove it. I will write a patch to enable the supply monitor when entering flash mode and submit a pull request when i tested it.

sskaug, 2016-12-08:
Thought this was the case. However, I would rather like to remove this from the code, as 1S is only very rarely used, and disabling the voltage monitor can give lots of strange problems, including flash corruption.

fishpepper, 2016-12-08:
What is the VRST voltage? When i understand it correctly it is 1.95V (efm8bb1-datasheet p16).
If it is really that low i think it is safe to remove this option completely.

Maybe there should be a small delay added between 
`orl VDM0CN, #080h ; Enable the VDD monitor`
and initialising RSTSRC as statet in the manual. The turn on time of the VDM0 monitor is 1us (p16 as well) ?

fishpepper, 2016-12-08:
I created a pull request that removes the one_s flag from all referenced files

sskaug, 2016-12-08:
There are a couple other changes in the pipe, particularly regarding multishot 32kHz, so there will likely be a rev16.6 soon, and I'll do this change then.