bitdump/BLHeli #203: support for EMAX Simon Series 30A

mrholub, 2016-12-10:
This ESC is not in the list of supported ESCs:

http://www.emaxmodel.com/esc/simonk-series/emax-simon-series-30a-for-muti-copter.html

Do you plan to support i in the future releases?

Kaeltis, 2017-08-23:
Would be interested in that as well, any new on the topic?

mrholub, 2017-08-23:
No there is apparently nothing new about this subject.
I've used ESCs with stock firmware in the end and configured these according to this howto:
https://www.youtube.com/watch?v=E6epG5E8gAQ
It works perfectly in my multicopter.

mrholub, 2017-08-23:
according to the Emax support, programming cards are not manufactured anymore for this model so I had to do this manually (can be annoying but it's doable).

fape, 2018-09-20:
see #254