bitdump/BLHeli #205: analogic pwm 0-5v instead of rc pwm

MakerDreams, 2016-12-12:
Hi all! 
that's a great work, i'd like to know if is possible by blheli, modify the rc pwm to take signal from a simple analogic pwm.
suggest?

sskaug, 2016-12-12:
I'm not quite sure what you mean by "analogic pwm", but in general any pwm input will be accepted, as long as it is one of the supported pulse widths or frequencies. Also 0-5V will work fine.

MakerDreams, 2016-12-12:
Thanks for the reply, i need to use the esc with a simple duty cycle pwm from 0 to 255 like this one: [Here](https://www.arduino.cc/en/uploads/Tutorial/pwm.gif)

And not with a classic servo pwm like this one: [Here](http://www.electrical4u.com/wp-content/uploads/2013/07/PWM-pulses-for-servo-motor.jpg)

Do you think is possible?
Where i need to modify the firmware of the esc?

JerryXing, 2016-12-14:
no problem, you modify? BLHeli modification is not necessary.

JerryXing, 2016-12-14:
But 100% PWM or 0% PWM, and signal loss can not distinguish.

MakerDreams, 2016-12-14:
@JerryXing  If any of you could tell me how to do that would be great!
That's great 0% to 100% duty cycle and I do not care if I lose the signal loss function.

Someone could make me or tell me how to make this change for the multistar 20A esc?

Thank you so much!

JerryXing, 2016-12-16:
@MakerDreams Modify int0_int()