bitdump/BLHeli #207: Features in BlHeli_Ssuite

JonnoSA, 2016-12-24:
Hi, please can you add an advanced option to BlHeli_S suite so that people like me can have access to all the settings that could be changed in the old BlHeliSuite?

I would really appreciate it if this could be added?

Many thanks

Jonno
[image]
[image]

yamahabest, 2017-01-02:
Is that even an option on BLHeli_S ESCs?

4712, 2017-01-02:
No, BLHeli_S does not support it.

brandonr49, 2017-01-12:
Has closed loop mode been removed in the change to BLHeli_S?
It's not in the GUI. Wondering if it will be added back later or if it's just not exposed.

Thanks!