bitdump/BLHeli #208: Compiling with SimplicityStudio

LightningStalker, 2017-01-02:
My path settings look fine, but attempting to compile I get this:

`$ ./MakeHexfiles.sh 
***** Batch file for BlHeli_S (from 4712)  v.2         *****
***** All Messages will be saved to MakeHex_Result.txt *****
***** Start compile with any key  - CTRL-C to abort    *****
compiling A_L_0
fixme:heap:RtlSetHeapInformation (nil) 1 (nil) 0 stub
fixme:heap:RtlSetHeapInformation (nil) 1 (nil) 0 stub
fixme:heap:RtlSetHeapInformation (nil) 1 (nil) 0 stub

ERROR: failed to run Ohx51
...
    FROM:    I:000020H   
    TO:      I:000028H   

*** ERROR L250: CODE SIZE LIMIT IN RESTRICTED VERSION EXCEEDED
    LIMIT:   0800H BYTES

Extended 8051/251 Object to Hex File Converter V1.43.0.0
COPYRIGHT KEIL ELEKTRONIK GmbH 2000 - 2012

*** ERROR: CANNOT OPEN FILE 'OUTPUT/A_L_0_REV16_5.OMF'

`

cturn, 2017-01-02:
I compile in windows, but I recognize a few errors..

[ ERROR: failed to run Ohx51 ]
- its looking for OHx51, I recall I had to download a few other files from another package?  
And linker fails, which results in this..
[ "*** ERROR: CANNOT OPEN FILE 'OUTPUT/A_L_0_REV16_5.OMF'" ]

[ *** ERROR L250: CODE SIZE LIMIT IN RESTRICTED VERSION EXCEEDED
LIMIT: 0800H BYTES ]
- again, I compile in windows, but at the end of my simplicity studio install, there was an option to license my toolchain, you actually do need a licensed version it seems.

LightningStalker, 2017-01-02:
Thanks a Keil  license fixed it.  Also I had to make a new project to get the ~/.config tree.

closed 2017-01-02 by LightningStalker