bitdump/BLHeli #209: BLHeli on Turnigy Plush 40A - Unresponsive to RC RX

Goby01, 2017-01-05:
I'm attempting to flash BLHeli on a set of Turnigy Plush 40A ESC's and make use of the bidirectional feature for ground based vehicle drive motors. As such I have no flight controller and intend to use the RC PWM as control signal. I've seen various success using simonK in this way...but that wont' work on the plush.

I've successfully (or so BLHeli Suite says) flashed the ESC with V14.8 Multi hex, I've made the following changes based on research and the manual.
-Enable PWM Input (on)
-Input Polarity (neg)
-PWM Frequency (low)
*Screen shot attached

Upon writing the new settings to the ESC and connecting the system to a motor and RX (LemonRX w/ DX5E TX) there is no beep or response from the controller. The unit is powered by a bench top supply and I have confirmed the TX and RX are bound. Calibration should not be required but I've gone through the steps and that didn't help. All electrical connections have been verified as well. 

I can't figure out whats going on, my motors are good, power is good etc etc

Any insights? I understand it may be PWM frequency related as I don't have a flight controller...but I don't see where (code or suite) to adjust that. Adjusting for dead bands do no make a difference either.

Input is very welcome, I've spent many hours researching and experimenting in order to figure this out.
I do not believe the esc is dead but have other fresh units I can attempt to work with if a solution is found. 

[image]

MarcoCanario, 2017-01-24:
Hello!

Same happened here! I already flashed a few of Plush 25A and 60A but with this ones I have troubles.
I have 6 Plush 40A, I already flashed 2 and nothing happened. I can read and flash again but when I power on they dont beep. I already tried connected to a Pixhawk, the original ones beep, the flashed nothing happened.
I also tried flash almost all versions since v8 to 14.9! 

This Plush have a different layout from the list! In the label of the Esc say ver4.2, they have already 5A Bec instead of 3A. I will put attached a photo of my Esc ( other component changed to another design was the big square with number 100 on it, I think is a coil because of the new Bec)
[image]
[image]

rotornoob, 2017-02-27:
Hello, me too it happenned the same with DYS XM 20A. I wanted to update my esc's with BL heli 14.9 (originally with bl heli 14.4 on it) using a naze 32 rev 5 with the passthrougth (i flashed betaflight 3.01 on my naze, i don't know if it does Something or not...?). During and the flashing, all seem to be find (or i so, i thought) but now, nothing happens when i plug my battery.  i am using "dys xm 20A firmware" .

   I tryed different versions of bl heli but nothing happened  (my motors are working, i tryed to make it spin with another esc and it worked) and before flashing my esc's, my motors was making sounds and was spinning normally but now, it won't do anything. 
   
   I can read my setup and flash my esc's normally too but it doesn't change anything.  I don't know what to do, have i done something wrong? because it was my first time using bl heli configarator i followed all the steps of that video below from Joshua Bardwell's YouTube channel:
[https://www.youtube.com/watch?annotation_id=annotation_2062529939&feature=iv&src_vid=YWLk4qcQcvw&v=mDNDpgKj0kA](url)

(ps: sorry for the poor grammar, i'm not fluent in English)