bitdump/BLHeli #211: Continuous operation

PeeceeDK, 2017-02-01:
Hi there!

Not necessarily an issue, but better safe than sorry.
Is the BLHeli or BLHeli_S capable of running continuously - i.e. from a powersupply? I'm thinking if any timers/counters overflows, etc.
Is it possible to program an unused GPIO pin to eRPM though PFM?

mikeller, 2017-02-02:
@PeeceeDK: Fyi: KISS 24A ESCs support eRPM feedback out of the box.

tracernz, 2017-02-02:
Original BLHeli can output eRPM on a GPIO (and some targets do by default), not sure about _S. KISS is a bit more complicated than GPIO pulses (ironic :stuck_out_tongue: ).

PeeceeDK, 2017-02-02:
I'm looking for a 6s version (stable 24V operation), making the KISS ESC not an option. So far i've been using the FVT Littlebee BLHeli_S 30A.

Can anyone tell me if it capable of running continuously? So far i've ran the ESC for ~3h without issues, but it's just recently i've begun to measure the RPM.