bitdump/BLHeli #213: BLHeli_S G-L-30 Multishot failed motor calibration starting from v16.5

phantom-8, 2017-02-13:
Tested on Racerstar 30A V1 ESCs.  Starting from BLHeli_S G-L-30 version 16.5, Betaflight failed ESC calibration when multishot is turned on.  Same problem on latest v16.6 firmware.  If downgraded back to v16.4 or below, ESC calibrated properly in Betaflight.

skorzo, 2017-06-29:
Problem confirmed with my Racerstar RS20A V1  (G-L-30) calibrating with multshot turned on.
Running Blheli_s 16.4 the are ESC calibrated properly in betaflight, but are not when running 16.5 and 16.6

Is this a G-L-30 issue or a general multishot issue like described here?
https://github.com/bitdump/BLHeli/issues/197

raoull14, 2017-08-01:
I have the same problem with my Racerstar 30A v1 ESCs and BLHeli_S G-L-30 v16.6 firmware.
ESCs keep on beeping "beep beep beep" and doesn't stop when I connect the battery with full throttle to proceed calibration on cleanflight and multishot actived. It's work with oneshot125 or oneshot42.