bitdump/BLHeli #214: Problem with motor timing on various BLHeli_S ESCs

maxsymba, 2017-02-15:
Hi
I have faced a serious problem, trying to build some new quads with modern hardware.
Had been tested such ESCc:
LittleBee 20A-S
LittleBee 30A-S
Aikon 30A-V2
Cikada 30A.

Some FCs with 6500 and 6000 gyro.

Problem description:
at the certain levels of throttle some oscillations (desync) of rotor is observed that creates the notable vibrations of motor and consequently - the vibration of all frame, generally by Yaw and, respectively, making the serious influence on the gyroscope sensor.
For sensitive gyroscopes, like 6500 this is insuperable problem, almost. From here and all attempts to use softmaunt, etc.
Usually such behavior of the motor speaks about incorrectly exposed timing. The attempts to correct it, having established suitable timing from all offered for the choice, didn't bring any significant result.
Any PIDs or other throttle controls with feedback processing didn't take a part in experiments. Only servotester, ESC and motor.

So. Have you any plans to improve a timing processing in BLHeli_S firmware? 
Autotiming (like YGE/YEP)? 
Any advice for actual firmware?

Best regards from Ukraine,
Max

sim-, 2017-02-15:
Hello! For which motor(s) and voltages are you seeing this problem? You say several brands, but which in particular?

It sounds odd that you are talking about this causing a problem for the gyros, since if it's really a desync, it's going to be really difficult for the flight controller to do much anyway. Are you sure you are not mixing up something like an interference pattern of vibrations which cause varying instability, as opposed to an actual loss of motor timing?

Maybe a video would help to make the problem easier to understand.

maxsymba, 2017-02-15:
I will made video tomorrow. There is night here.
Motors (all on 4S-16.5V): 
* EMAX RS2205 - 2300KV,
* EMAX RS2205-S - 2300KV
* T-MOTOR F40-2300KV, 
* T-MOTOR F40-II - 2400KV,
* ZMX Fusion X25-2206 - 2300KV
* BrotherHobby T2-2206 - 2300KV
* Lumenier RX2206-11 2350Kv
I repeat - eventually only servo-tester, ESC and motor. 
I started hobby from the helicopters, so I got used to setup of proper timing and I know a symptoms when the timing is wrong.
I tryed a KISS-24A. There are some moments of desync too. Even the count of them is much more for all range of throttle. But they much smaller than on BLHeli_S. Therefore this problem isn't so actual for them.

cturn, 2017-02-16:
Not an ESC/Motor problem, what you have there Is a flight controller sensitivity/mounting issue.

But I'll play along.  You say you can reproduce the issue with a servo tester?  If so at what throttle range/conditions are you having this problem?  

Your reference to Heli's and YEP/YGE esc's makes me think you may be referring to output pwm frequency?  Which is adjustable in most Heli esc's but this function is handled in hardware on all blheli_s esc's and is 24khz and not adjustable.

maxsymba, 2017-02-16:
The most expressive desync - at the level of throttle is slightly lower than the center (1500-).
I will make a video and will try to show all conditions and circumstances.

odixxx, 2017-02-16:
"The most expressive desync - at the level of throttle is slightly lower than the center (1500-)."
I have exactly the problem!
on two different quadrocopters
F4 - 6000 gyro 8kHz

ESC Racerstar MS35A (A-H-20)
DYS SE2205 2300KV

ESC Racerstar RS30A LiteS (G-L-30)
Racerstar BR2205S 2300KV

cturn, 2017-02-16:
Odixxx
This sounds like a harmonic issue, where that particular motor/esc combo creates some kind of resonance with the frame, or something else that the gyro is than picking up.  A BB log with the noise/vibration frequency graph enabled should pick this up, and allow you to filter out the noise/vibration using one of the notch filters.  

I'm trying to be as helpful as possible but this REALLY sounds more like a FC tuning/dampening issue, not a blheli issue.

odixxx, 2017-02-16:
Tomorrow I will write down the black box
and try to reduce gyro_lowpass, dterm_lowpass

maxsymba, 2017-02-17:
Hi.
I made a video:
https://www.youtube.com/watch?v=KUdFO9vlJKs
There are balanced F40 with no sideplay in bearings, LittleBee 30A-s on 16.5 (but 16.1-16.6 have same troubles) and servotester.
There are very outstanding oscillations at least on 1425-1438. Also You can hear some oscillations on other levels of throttle.
Also it is worth noticing that this oscillations heats up the motors very much.
So - no gyro's feedback, no disbalance vibrations - only timing's issue.
Other motors and ESCs brings the same results.
I also tested non-S LittleBee 30A - same result.
I am sure - You can easily repeat these results yourself.

sjh2100, 2017-02-17:
Steffen Skaug and every e-pal
**************please note the word beside and in【】
; Fast calculation (Comm_Period4x_H less than 2)
calc_next_comm_timing_fast:			
	; Calculate new commutation time
	mov	Temp3, Comm_Period4x_L	; Comm_Period4x(-l-h) holds the time of 4 commutations
	mov	Temp4, Comm_Period4x_H
	mov	A, Temp4				; Divide by 2 4 times
	swap	A
	mov	Temp7, A			;i think temp7=H_low_high
	mov	A, Temp3
	swap A
	anl	A, #0Fh				;i know A=L_0000_high
	orl	A, Temp7			;why 【orl】 here???? 

**************
; in Measure lipo cells 

; Add the programmed number of 0.1V (or 3.125% increments)
	mov	Temp3, Bit_Access		; Load programmed limit (Bit_Access has Pgm_Low_Voltage_Lim)
	dec	Temp3				;i think 【mov a,temp】is absent
	jnz	measure_lipo_limit_on	; Is low voltage limiting on?
**************

 Set pwm limit low rpm 
; 
; No assumptions
;
; Sets power limit for low rpms and disables demag for low rpms
;

	mov	A, Comm_Period4x_H
	jz	set_pwm_limit_low_rpm_exit		; Avoid divide by zero

	mov	A, #255		; Divide 255 by Comm_Period4x_H，【why Divide 255 by Comm_Period4x_H？】

**************
; Second governor routine - calculate governor proportional error 
calc_governor_prop_error:
【in control pinrciple or automatic control system，
the input is set point ，

and the output is variable controlled bysystem ，

so error=input-output.if Gov_Target_L is output,Governor_Req_Pwm is input,

i think it should like error=Governor_Req_Pwm-Gov_Target_L】
IF MODE == 2	; Multi
	; Calculate error
	clr	C
	mov	A, Gov_Target_L
	subb	A, Governor_Req_Pwm		; Gov_Target_H Gov_Target_L- Governor_Req_Pwm
	mov	Temp1, A			;why not【current pwm- Governor_Req_Pwm？】

**************

; Calculate governor routines
; No assumptions
; 
; Governs headspeed  based upon the Comm_Period4x variable and pwm
; The governor task is split into several routines in order to distribute processing time
;
;**** 
; First governor routine - calculate governor target

governor_activate:
	mov	Temp1, #Pgm_Gov_Mode		; Store gov mode 
	mov	A, @Temp1
	mov	Temp5, A
	setb	Flags1.GOV_ACTIVE
	mov	A, Requested_Pwm			; Load requested pwm
	mov	Governor_Req_Pwm, A			; Set governor requested pwm   pwm
	; Calculate comm period target 2*(51000/Requested_Pwm)【what means this “2*(51000/Requested_Pwm)”】
my emal 【sjh2100#sohu.com】，hope we can discuss that

maxsymba, 2017-02-20:
Any updates?
BTW: At present I see flight videos  from other pilots and I clearly hear this oscillations!

maxsymba, 2017-03-09:
Yesterday I tryed same motor on YEP 40A with autotiming. Smooth as silk. 
Q.E.D.
Can be possible to fast implement a numerical definition for timing-parameter (0,1,2...30deg)? And to leave presets (low/med/high) only as configurator's feature? Though I think that this problem is related not so much to the wrong choice of timing, how many to some bugs in a firmware code.
There is one more hypothesis that maybe with increase of RPM some correction of timing is necessary.

sim-, 2017-03-09:
Q.E.D? That to me would just stand out as something good to compare on the oscilloscope, not Q.E.D. ;)

Are you comparing YEP 40A without complementary PWM ("damped light") mode versus BL-Heli with it? It is must easier to hit interference patterns between the PWM frequency and the commutation frequency when complementary / damped mode is enabled.

There are also various ways to implement the PWM clock. Some ESCs reset it at each commutation step, for example (which can often make the problem worse), and it can sometimes help to significantly increase the PWM frequency.

Finally, the motor design versus the trapezoidal drive technique will cause some torque ripple which will show up more drastically with complementary PWM since it will actually cause braking and acceleration with each PWM cycle. Since you are running "unloaded", the effect will be much more apparent than with some mass attached.

maxsymba, 2017-03-09:
Listen, the good test with YEP (even with active brake) only prooves that this is not a motor's issue. I don't belittle achievements of BLHeli at all. But the declared problem is a PROBLEM! And it doesn't depend on a brand of the ESC or the motor. The only correlation - if more powerful motor (strong magnets, etc), then problem is more obvious.
And it would be silly just not to pay attention to existence of this problem. Because IMHO this problem spoils life to a lot of people in a varying degree.

sskaug, 2017-03-09:
This is most likely harmonics of the commutation frequency aligning with the motor pwm frequency. The worst one is at about 240k erpm (for BLHeli_S running 24kHz pwm). Most motors do not reach this rpm in most common system. But smaller motors in low load systems can. Also for 120k and 80k erpm there are points where harmonics align, but these are smaller and somewhat masked by FW.

Probably the best and cleanest way around this is to increase motor pwm frequency. For BLHeli_32 it will be possible to run up to 48kHz, which in initial testing is running very smoothly.

maxsymba, 2017-03-09:
I don't know, why you said "low load system" because this oscillations presents both with props and without. 
If in the loaded state (with props and even in flight) the problem disappeared - questions wouldn't be. 
Lads, I hope you will fix this shit somehow....

P.S. I remind the question about numerical definition for timing property.

maxsymba, 2017-03-14:
The problem is also reproducable through the DShot((

maxsymba, 2017-03-21:
Can somebody provide some guid about BLHeli functional principles and also - howto "to compile firmware", "build testing stand", etc? I see that so far very few people pay attention to this issue. Therefore if I have free time, I will try to investigate it by self.

Otakee, 2017-04-09:
I have a similar problem, 

See on blackbox screenshot, 

[image]

chapimp, 2017-10-02:
I got the same issue, but only on motors 1 and 2.
motors 3 and 4 run smooth.
Im using the Alpha AIO FC from [getfpv.com](https://www.getfpv.com/lumenier-alpha-aio-flight-controller-f4-fc-4x-30a-blheli-32-esc-osd-pdb-curr.html?gclid=Cj0KCQjwgb3OBRDNARIsAOyZbxCXj1STsi5rpun_7VjOwxbGrYrbtv9fQ1jhsmQ0xpBDTMHTTehny14aAswHEALw_wcB)

I'm sure is not the **gyro** cause Im testing on the **Motors tab**, in **betaflight**, I did even try motor test on **BLheli suite 32**. and the same 1,2 motors do a grinding noise at specific throttle levels.

maxsymba, 2017-10-02:
Did you raise the frequency to 48kHz on BLHeli32?

chapimp, 2017-10-02:
@maxsymba 
I changed all settings in BLheli and no difference.
I thought it was my  motors, so I swapped them and the same issue on the 1,2 motors.
It could be a defective board, I'm exchanging it, hoping it was a bad board.

tekno7620, 2017-10-19:
I have the same problem non many motors with many different Blheli_S esc....at mid throttle the motor makes alot of vibration, this vibration go to the pid loop, and the quad fly bad....any solution? Thanks

AdrY89, 2017-11-20:
Hello guys, I have changed the esc on my quad...
And I found the problem that the motor 4 do not spin at level mode regular at low throttle...
In acro go well
DSHOT 600
RACERSTAR 30A LITE S
MOTOR RACERSTAR 2306 2300KV
FC MATEK F4
I believed that it was the motor or the esc... but then I changed the esc and the motor and the problem there is... so I figure out with multishot... the motor goes fine

Qopter, 2018-01-25:
May I dig up the old thread again. I have the same problem. There is a interference with certain throttle positions. Here are two pictures that visualize that. I ramped up Motor 4 in the Motor Tab and recorded it with Blackbox. There is no PID loop involved. At about 34% throttle you can see the interference on gyro yaw. 
[image]

The second picture shows the associated spectrogram. The line with the ramp is the motor revolution and ramps up from 50 to 300 Hz. The V-line is the interference. The slope of the interference is 14 times higher than the ramp. The interference line crosses zero frequency at about 11.3 sec. There are other disturbances in the spectrogram, the neighboring lines to the main lines. They are not from the ESC but from the FC. You can ignore them.
[image]

My question to the devs: What exactly is happening? Is there already a solution?
My BLheli_S version is 16.6

AdrY89, 2018-01-25:
In my case, the problem was a little piece of tin on joint of esc 3 - 4 that made the problem… of bad signal for esc 4.
Then for the linear current on the quadcopter, the matek on old F4 AIO ( that use the 6500 imu very sensitive) suggest to mount or 2 cap ON esc 3 and 4 or 4 cap on esc 1 2 3 4.