bitdump/BLHeli #216: Flycolor Fairy 30A - Which BLHEeli_S hex?

RyanSA, 2017-02-23:
As the Flycolor Fairy 30A's run the F850 MCU they can run both BLHeli and BLHeli_S firmware.

Which BLHeli_S hex would these ECS's be able to run? I've read that A_L_70 would work?

I'm running the v1's of the Fairy's - pictures can be found on this thread - https://www.rcgroups.com/forums/showthread.php?2834476-Anyone-help-me-identify-these-ESCs

Thanks.

RyanSA, 2017-03-13:
Has anyone tested these on A_L_70?

mfur, 2017-06-25:
Bump, In am also interested. This page says C-H-15: https://github.com/blheli-configurator/blheli-configurator/wiki/BLHeli_S-Firmware-Knowledge-Base

If anyone tested those ESC with BLHeli-S I would appreciate the feedback (also about potential capacitor removal).

personmans, 2017-11-23:
C_H_15 gives a message saying it is not the right type of target.
A_L_70 Does not work; Instantly burned 4 ESCs on power up -- I should have limited the current, lesson learned.