bitdump/BLHeli #217: Roll of Death

simonbethke, 2017-02-23:
This is an issue, that was reported recently like 4 times in a row in the german forum fpv-racing-forum.de.

The catchy name "Roll of Death" shall describe the issue that results in this experience:
_A pilot tries to do a roll. The roll works perfectly, but when trying to stop it, the copter starts to accellerate the roll. This always results in a crash._

I already discussed this a few times on the betaflight slack group and we found that this is not caused by the FC. 

**Observation:** [BlackMarlin12](http://fpv-racing-forum.de/user/1217-blackmarlin12/) reported, that the issue did not occur again after switching off damped light.

**Theory:** When having throttle at zero during a roll, the props are on idle speed. I think it could happen that the props on one side of the copter eventually start spinning backward due to the airstream.
Furthermore, I think that BLHeli might be not prepared for backward spinning props if no 3D setup is used. 

However, either active braking (which would support the damped-light observation) or the command to spin up the prop in the correct direction will result in the prop spinning up in the wrong direction.
Ofcourse this will lead to a PID-Feedbackloop which will max the speed of this prop in milliseconds.

I already explained this theory to several people but nobody was able to give good pro or con arguments. Just all people gave it the mythbusters "Plausible" tag.

**Material:**
- Here is a screenshot of BLHeli Suite: http://fpv-racing-forum.de/attachment/6482-escsettings-png/
- Video of the roll-of-death:  https://www.youtube.com/watch?v=MQ7LZgack8E
- Betaflight Blackbox log: https://drive.google.com/open?id=0B9ZAczyQPjrYckVqOUJpdzZyX0U
- Betaflight Blackbox log2: https://drive.google.com/open?id=0B9ZAczyQPjrYUl95VVFaNWNpTzQ

Both blackbox logs have holes that seem to be caused by bad writing speeds of the SD-Card. I will provide better logs as soon as I have them.

simonbethke, 2017-02-23:
Oh no. I posted wrong Material. I will fix it now.

simonbethke, 2017-02-23:
Material fixed.

BlackMarlin12, 2017-02-23:
Hi, I'm affected by te Roll of Death. It would be nice if there's someone who can help us.

simonbethke, 2017-02-26:
Dear developer of blheli, what is your prefered channel for communicating issues related to blheli. 
Being a Softwaredeveloper myself, I very well understand that you might not want to jump on this immediately. However, I would really like to read at least a first oppinion from you.

sskaug, 2017-02-26:
Generally these issues are better covered in forums like RCG.
And generally, increasing min_throttle, increasing timing or increasing demag comp helps against such issues. All these things have been widely discussed in forums.
And unless you have a badly wound motor or an ESC that tends to reset in flight, one or more of the above should fix your issue.

simonbethke, 2017-02-26:
Thanks for the quick Response now. So

simonbethke, 2017-02-26:
The cancel Button instead closes The issue and posts The comment...

cubitusclaudius, 2019-03-01:
Interesting. I'm just having such problems, and was wondering if dshot_idle should be changed (dshot600 in my case). This value doesn't exist no more under configuration tab, but is still in the CLI. I raised it up and it deemed to get me rid of the problem :
I first raised it to 6.5 (it was 5.5%) and tried a lot of flips and rolls without problem, except on my last lipo where it came back again. so I raised ti to 7.0% and till now no more desync... but I just flew 2 lipos since then...