bitdump/BLHeli #226: compiling BLHeli_S

johnty, 2017-03-21:
first of all, thanks for the awesome work on this project!

after messing around a bit i managed to compile it using the batch file, but the process was not entirely obvious or officially documented, as far as i can see. the main challenge was choosing the right toolkits, and licensing issues surrounding the keil compiler that restricted output size by default.

unless i totally missed it, a short description in a readme with a link to [this page](http://pages.silabs.com/lp-keil-pk51.html) would probably save new users quite a bit of time. perhaps i simply didn't look in the right place and this information already exists somewhere?

likewise, for the older SiLabs version, simply pointing to the install of the 8-bit toolkit package as opposed to Simplicity Studio, which do not install the SDK files in the expected paths of the make file - i.e. root/MCU/...),

once again, apologies if this info is documented already and i simply missed it! :)

finally, if there are other (legal or otherwise) reasons for not having the information easily accessible, i would understand as well. but otherwise, hopefully this suggestion may result in some further documentation that may prove to be useful to others! :)

chatch15117, 2017-03-29:
hi, there's no licensing issues with this, it's very straightforward. 
[1. Download SSS v3 ](http://www.silabs.com/products/development-tools/software/simplicity-studio-version3)
2. install
3. select package 8051 for EFM8 and EFM32
4. register KEIL for free as described in the installer
5. run makehexfiles

johnty, 2017-03-30:
Great, thanks! That short list is exactly what would be quite helpful, as you can see I went through a somewhat roundabout way and eventually got it to work, having tried SSS v4 at first since its the default one (which has different folder structure for the headers etc), and then running into various hiccups along the way.

Link774, 2017-04-13:
I'm having some trouble figuring this out as well. I couldn't find Package 8051 for EFM8 in the install, but I found EFM8/C8051 8-Bit Products under the Product Family tab. Does that work? Also, it never asked for my KEIL registration code. I registered using the link johnty posted, but it is unclear where to put the registration code. The registration page linked to a document that supposedly contains instructions, but I can't figure out the step "Open the Keil µVision4 IDE from the installation directory with administrative privileges." Any ideas? Thanks!

audiomodule, 2017-05-24:
Do you need **BLHELI ESC** ? 
#TATTU [BLHELI_S ESC 30A](url =http://www.genstattu.com/ta-esc-30a-blheli-s.html  blheli_s esc  30a) 15 Off!
Who likes to try our New Arrival #Tattu ESCs for FPV? We have both whole sets and single one. 
Let me see the interested hands!  Free shipping over $49.99 to USA
USA 15% off : http://www.genstattu.com/ta-esc-30a-4pcs.html
Germany: 
http://www.gensace.de/tattu-esc-30a-blheli-s-2-5s-w-dshot-no-bec-prototype-1153.html

chatch15117, 2017-05-24:
Buy a 45A version for cheaper than Tattu, also higher quality. http://realfpv.com