bitdump/BLHeli #228: Cobra CP2207 2600KV

bonhardt, 2017-04-04:
1 / 5 times from throttle 1280~ fast increase to around 1750~ the motor gives weird noise
from tests it happend not always on the same motor.

resulting = roll of death crash

i guess it is some kind of sync issue? blheli_s 16.6 default settings

Otakee, 2017-04-09:
I have a same issue on F40 PRO 2600kv with Flip 32 F4 and little bee 35A on DSHOT protocol. 

I think a problem is on the last version 16.65

robschii, 2017-04-13:
i have 16.6 and dshot 600  on racerstar30V2 and spf3.  

 bf3.1.7 + BLheli16.6      quad flips automaticly at full throttle    
bf3.1.6 abnd 3.1.5   + blheli 16.6    one motor sound terrible after 2 minutes of flight with punchouts and 100% throttle    and has now power from this moment

bonhardt, 2017-04-14:
After discussing with many people the problem is probably this new motors with stronger magnets generating increased noise back to the system resulting esc can reboot or act not normal.

DieHertz told me i should add some capacitator near where the battery connects to the PDB or each ESC.

At the moment i putted 470uF 35V low ESR capacitator, so far the issue is gone, but didnt had enough time to test properly.

robschii, 2017-04-15:
i have already one cap     but i fly this quad over one year  but nuow i have such problems. i had switched the motor1 + esc  with motor 3  but the same problem with motor 1