bitdump/BLHeli #230: Question: Is FOC the Next Big Thing in smooth & efficient motor controls?

RipperDrone, 2017-04-19:
@sskaug I just did some research on FOC motor controls as opposed to trapezoid or sinusoidal. It seems that one particular problem with today's noise sensitive gyros can be addressed effectively: FOC provides uniform, almost ripple-free torque curves which might enable us to finally use motor/frame noise sensitive faster gyros without the need for softmounting, thereby compromising PID loop dynamics.

http://www.atmel.com/Images/doc32126.pdf (edited)

joner [7:52 AM] 
repli nice document

repli [7:56 AM] 
With the advent of STMF7, could one imagine to pack all the number crunching parts of FOC algorithms on the FC itself and just have the ESCs handle the motor power loop control? Might need bidi communication btw FC and ESC, though - to feedback current sensing...

Below are some resources I have identified. What do you think of it?

https://www.roboteq.com/index.php/applications/100-how-to/359-field-oriented-control-foc-made-ultra-simple (edited)

OlliW has investigated the FOC motor control option for his STorM32 Gimbal project already: http://www.olliw.eu/2017/field-oriented-control/ - maybe he can be of help in case we want to implement

FOC-capable ESC reference designs seem to be existing - here is one from Infineon. Detailed explanations included in .ppt: http://www.infineon.com/dgdl/Infineon-Application-Motor_Control-Drone_Electronic_Speed_Controller_ESC-TR-v01_00-EN.pdf?fileId=5546d462580663ef015843a229fe54ea

TI is on the FOC spree as well: http://www.ti.com/lit/ug/tiducf1/tiducf1.pdf

tracernz, 2017-04-19:
It's not really the "next big thing", it's been a thing for a long time. Already some relatively old open-source implementations e.g. http://vedder.se/2015/01/vesc-open-source-esc/

sskaug, 2017-04-20:
I once tried an FOC ESC, the EMAX Windtalker. But I didn't have the motor that it was set up to pair with, and it ran very poorly. How to solve the issue of matching motor to ESC parameters?

RipperDrone, 2017-04-20:
Since you have introduced an authentication system for your BLHeli_S firmware anyway, couldn't this be a route to authenticate motor IDs which contain all required information to drive them FOC properly?

One might even think about bundling motors and ESCs, like Racerstar, EMAX, ... are striving to push for anyway.

Maybe QR codes printed on the motors and/or motor boxes could also help authenticating the FOC parameters required for the motors...

tecmansun, 2017-04-22:
As I know, motor param. can be indentified automatically in order to match
different PMSM.

LiangAndrea, 2017-05-09:
Hello@sskaug I am from FrSky, my colleague of R&D department want to contact with you, please send me the mail (haoliang.huang@frsky-rc.com) if you think this is okay. Thank you!

controlflow, 2018-01-23:
Interesting, just announced DJI Mavic Air drone mentions FOC ESCs in marketing materials: https://www.dji.com/mavic-air?utm_source=newsletter&utm_medium=edm&utm_campaign=mavic-air&utm_content=release&sc_src=email_1694741&sc_eh=43a80757fc71f2821&sc_llid=81821&sc_lid=69868312&sc_uid=SXrhMEoTqb

Changliu52, 2018-11-27:
I think foc is really the key to make the blheli be used for much wider applications like ground rovers or even legged robots. I’m not sure how keen is the community to make its possible here? Will the hardware be significantly different?

jkflying, 2019-03-03:
From what I understand FOC needs a voltage and a current reading from each phase. There is also a considerable amount of math that needs to be done to determine the phase angle, I'm not sure the accuracy required but I suspect an FPU would be useful at least in the beginning, until it can be optimized.