bitdump/BLHeli #232: Beep frequencies

varkokonyi, 2017-05-06:
Hi all. Is there a way to convert

	mov	Temp3, #44
	mov Temp4, #125
	mov Temp5, #1
	call music

(=ESC beeps) to a sound frequency?

varkokonyi, 2017-05-06:
(I know that temp3 is the length)

jackbnimbl, 2017-05-11:
I mapped out the notes while creating the Mario Underground theme.  I don't know how to request it be added to the source code but I think this is all someone really needs:
;**** **** **** **** **** **** **** **** **** **** **** **** **** **** **** **** ****;
;								 		     ;
;		Music Routines					                     ;
;										     ;
; Temp5 values range from 1-256, Temp4 value ranges from 1-4+                        ;
;										     ;
; Temp5 notes for Temp4=1 							     ;
; 1:b6 25:a6 50:g6 65:fs6 100:e6 125:d6 150:cs6 175:c6 200:b5 256:a5		     ;
; 										     ;
; Temp5 notes for Temp4=2							     ;
; 1:b5 25:as5 50:a5 75:g5 100:g5 125:fa5 150:e5 175:f5 200:e5 256:ds		     ;
;										     ;
; Temp 5 notes for Temp4=3							     ;
; 1:e5 25:e5 50:ds5 75:d5 100:d5 125:cs5 150:cm5 175:c5 200:b4 225:as4 256:as4	     ;
;										     ;
; Temp 5 notes for Temp4=4							     ;
; 1:b4 25:as4 50:as4 75:as4 100:a4 125:a4 150:gs4 175:gs4 200:g4 225fs4: 256:f4	     ;
; 										     ;
;**** **** **** **** **** **** **** **** **** **** **** **** **** **** **** **** ****;

varkokonyi, 2017-05-11:
Yes, I'm working on the Nokia tune. Thank you, this helps a lot. I will try to submit a pr to add this as a comment to the music section. I will add something like "-by jacknimbl"