bitdump/BLHeli #237: ProShot?

ctzsnooze, 2017-05-22:
Hi Steffen - just curious on your thoughts about Tony's ProShot protocol, in terms of technical merit in relation to DShot?  It seems to me to be very good; I'd love to see it in BLHeli.

cturn, 2017-05-22:
have a link to info on this "pro-shot"?  I'd like to check it out.

ctzsnooze, 2017-05-22:
https://www.rcgroups.com/forums/showthread.php?2879092-%2A%2AIdea%2A%2A-ProShot-Another-digital-protocol

On page 2 are videos of the code driving ESCs.  Now most people are arguing about the name.

Intent is that it allows data rates of DShot 600 with greater reliability and less processing overhead in the ESC MCU and in the FC.  Boris has said the FC coding side is not complex.

The developer (Ub3r in rcroups) is a professional custom ESC firmware and hardware designer with *lots* of ESC related experience.

mikeller, 2017-10-15:
@sskaug, @cturn: Is there any progress in adding ProShot to BLHeli? I think this would be a worthwile addition (cutting down on interrupt load / DMA requirements).

sskaug, 2017-10-15:
I see that it is now included in BF 3.2. Probably it will be added to BLHeli_32, which should be quite easy.

richard-scott, 2018-01-15:
Was this ever added to Blheli_32?

sskaug, 2018-01-15:
It is implemented in BLHeli_32. Not yet in official release code, at the moment just in the test code to be found here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM