bitdump/BLHeli #243: Quick help with MP6530 driver

Engineereric, 2017-06-26:
I doing a new design with a MP6530 driver. Perfect for ESC's because it is designed for 3-phase Brushless motor MOSFET drive.
The issue is unlike many of the basic ESC's that have separate drive for the low side and High side MOSFET's this just needs an enable signal and a PWM signal for each phase.
Its easy for me to modify the main code for BLHeli but I was wondering if there was already a configuration that used this style of driver?
Thanks in advance.
Eric

SandyZhao17, 2017-07-07:
I'm doing this new design also, looking forward to the reply.

Sandy

archfiend, 2018-10-08:
Did you guys make on progress on this?  I'm interested as well.

Engineereric, 2018-10-08:
No turns out another manufacturer already uses it and I was able to re-code that version.
I cant remeber the letter off the top of my head as that was over a year ago. However I can confirm the ESC I designed worked well with the configuration for that driver. I just had to manage the pins correctly.
My original post had been hoping that some one would have been able to tell me what chips each version made use off. In the end it was just luck I found a blown ESC with the very driver chip on it I was using. I posted a YouTube video of me running a motor at 86kRPM with the prototype.

archfiend, 2018-10-09:
Thanks for getting back to me on this, we're tearing our hair out trying to get this driver working.  I've looked through every configuration and none of them seem to be correct.  Clearly I'm missing something.

Are you able to share your youtube link?  

Do you recall any more details about the ESC type that used this driver?  

Thanks in advance, we are in a real jam.

sskaug, 2018-10-09:
This should be real easy. BLHeli and BLHeli_S both support EN/PWM style drivers like the MP6530. Just set deadime to zero (NFETON_DELAY/PFETON_DELAY in BLHeli).

Engineereric, 2018-10-09:
If memory serves me you need to reverse the drive PWM and high side drive, I'll need to find my original files but you will need to configure for your pin allocation.
YouTube is at:-
https://www.youtube.com/watch?v=TfL3JQ4f83I

archfiend, 2018-10-18:
Thanks @sskaug and @Engineereric for all your help.  We have a build working now, we modified the 'S' config to match our pinning, polarity and driver.