bitdump/BLHeli #244: What happened to the Low Voltage Cutoff setting?

traxxi2003, 2017-07-02:
Hi, Im working on a quadrotor project that is using a custom build Lithium Ion battery pack in a 6S6P configuration. This battery can be safely discharged down to 15.0V. After flashing my Hobbywing Xrotor esc's with BLHeli 14.4 the "LVC" is not longer visible. Does that means I can no longer disable that feature? What is the default cutoff voltage on later firmware versions? after reviewing previous blheli firmware versions it appears that after version 13.2 the "LVC" setting was removed. How do I make sure that the esc's wont cutoff my motors when discharging the battery passed 3.7v per cell?

nitr0123, 2017-08-26:
I don't think they cut off at all now so you're fine

maroquio, 2019-04-14:
Hi, traxxi2003. Did you handle it? I have the same problem. I use Li-Ion batteries and the can cuttoff at 2.5V per cell, but ESC is cutting at 3V per cell.

traxxi2003, 2019-04-15:
Hi maroquio, after BLHeli 13.2 Firmware the option to change, turn on/off the voltage cutoff was removed from the software. This in turn allows the ESC to continue to operate until the battery is fully drained. With my setup I have flown my quadcopter down to 2.5v per cell and the esc do not cutoff. Not sure which flight controller you are using but make sure that the FC is not the reason for your setup not working below 3.0v per cell. Hope this can help!!

traxxi2003, 2019-04-15:
[image]
[image]
Look at these pictures; notice how on the newer version of blheli the option setup “voltage cutoff “ has been removed? That means that the ESC’s will continue to operate until the battery is fully drained.

maroquio, 2019-04-15:
Hi!
Thanks for your answer. I'm trying Pixracer R14 and Kakute F7 AIO. Both works okay, but Kakute is lighter and more expansible. I discovered that now only BLHeli32 have the option to disable the LVC. ESCs that use BLHeli_S can't be modified. Some come with LVC from factory, some not. Up to now, I know that HAKRC BLHeli_S ESCs come with LVC from factory, and Lumenier BLHeli_S ESCs don't. I have 4 more models of ESC on the way for testing. Lets see what we discover. 
Regards!

a146860, 2019-08-21:
BLHeli_S   There are  Low Voltage Cutoff？

Howie9600, 2019-08-25:
i am seeing the same issue.  on a 2s battery the esc is cutting too early.  i want to disable this.  any ideas?

maroquio, 2019-08-25:
There is a Chinese guy on YouTube that did it bypassing a resistor on the PCB. I don’t know electronics, but this is a way to do it. By firmware, only in BLHeli32 supports this configuration by software.
Regards,
Ricardo Maroquio

Howie9600, 2019-08-25:
So BLHELI 32 allows you to turn it off but the ESC still shuts down if you go below 7v.  I am using 18650 cells and the voltage drops to 5.5 volts.  But I can't get anywhere near that as the ESC turns off.  No matter what I do in the software it still happens.

Do you have a link for the Chinese guy?

maroquio, 2019-08-25:
Sorry, I don’t have the link. He claims for world record flight time in a under 250g quad rotor. That’s how I found him... I’m developing under 250g quadrotor too and I’m using 2S 18650 battery. If you use a 1-2S ESC, it use to discharge below 5.5V. I’m using one from Cicada, 7A/10A 4-in-1.
Regards,
Ricardo Maroquio

Howie9600, 2019-08-25:
Thanks for that.  Do you get any over heating issues?  I used a 5a teeny 4 in 1 and it basically melted.  I'm only pulling 3 amps in hover and it over heated in under a min.  Does that one your using over heat?

a146860, 2019-08-25:
May no longer have a low pressure cutoff

maroquio, 2019-08-25:
I tried this one from Teeny too. It’s really a crap. If you compare the capacitor on the PCB, you can see a big difference from the others. There is another one, Favourite Littlebee 10A, that worked fine for me too. I’ve tested 13 models of ESC for my project. Actually, I’m using Lumenier 20A 4-in-1 because it has a 5V 2A BEC that is very useful for me. It cuts off at 5.8V, what is okay for me.
Regards,
Ricardo Maroquio

Howie9600, 2019-10-13:
Any more news?  Still struggling to find an ESC that will run down to 5v.

gary-w-johnson, 2019-11-12:
Found Mod to 4 in 1 esc
Check https://www.youtube.com/watch?v=E5RKxA2rEC8
 https://drive.google.com/open?id=13O36xw_DISa1GcLIJBK88Ees2azNxgc5

Normal 2s-4s 4in1 ESC like kingkong 20Ax4 working voltage is between 26v to 6.2v. In my case, I modify my 4in1 ESC to operate at lowest voltage down to 4.2v (Tips : Bypass 12V regulator and diode).  Vbat -> diode -> 1st regulator -> 2nd regulator and halfbridge.  The first 12v regulator(sot-89) connect to the second regulator 3.3v(for MCU) and half bridge NMOS driver(for NMOS). If battery voltage = 6.2v, after diode (-0.3v), after first 12v regulater (-1.7v) then output 4.2v to the second 3.3v regulator and half bridge NMOS driver. At this low voltage half bridge won't work at all. Short diode and 12V regulator in and out will solve this problem and give more 2.0v margin. ESC after mod only can be operated between 4.2v to 15v (3s max)😁.  4s may be dangerous.