bitdump/BLHeli #247: blheli for non-RC use !

Hrumque, 2017-07-15:
I suspect there would be a lot of happy people if there was a firmware version that was as simplified as non-RC.
For example, I am using bldc engine together with blheli-ESC as a drive in a milling machine, and modified battery drill (overkill, I know). It would be great if there were no unnecessary hissing beeping and arming actions after each power on, just like with brushed motor. 

So - the total absence of arming procedure, just after the power supply - the motor turns into a pulse of PWM signal. Or even better - analog value (if the processor can measure the supply voltage and temperature = the voltage on some NTC sensor, then you could solder the cable and measure the voltage from the potentiometer to adjust the rotation (then the pwm connector could remain for blheli_suite programming purposes).
Or setting the rotation through some serial interface, even software uart-ttl on the pwm-servo line, and simple command type "S1000" = 1000rpm, people from the game with arduino would be delighted;)

hugokernel, 2017-07-20:
Hi,
I have exactly the same need.
I want to use a micro brushless motor for non-RC use and I do not need arming and just want to have a motor who turn at XX rpm or from an external information (PWM, voltage, etc...) immediately after power on.

hugokernel, 2018-05-20:
No one else interested by this feature?

danyaPostfactum, 2019-10-24:
It's quite easy. Just remove initial beeps, arming check and calibration procedure from the code. I need almost the same thing, but I'd like to control a motor by a potentiometer (one direction). It is easy to remove the code lines, but I'm  not sure I will master a potentiometer control.