bitdump/BLHeli #249: Main and Tail support in BLHeli_S

arrowcircle, 2017-07-26:
Hey!

It looks like Main and Tail motors support is dropped from BLHeli_S.

New ESCs with _S firmware works much better and smoother as well as lightweight and smaller in size. Support for helis will improve performance with new ESC greatly.

Is there are any plans on reviving support for BLHeli_S?
How much work needed to make it? Or is it possible to hack and make a build for exact ESC?

berndpfrommer, 2018-04-15:
I'd love to be able to run a governor on new BLHeli_S hardware. There seem to be currently no "Main" and "Tail" versions (I need only "Main"). Are there any plans to support that?

jj-uk, 2018-07-11:
Another vote for this. I need main support for the A-H-70 escs ( DYS XSC 20A ).

It looks like BLHeli no longer supports, erm, heli's !

nen77, 2020-04-28:
Hey..any solution for new esc to flash with tail firmware ?