bitdump/BLHeli #251: BLHeli_S: 4way-if not finding ESC (C2 Interface)

swilkens, 2017-07-29:
I'm trying to connect to an ESC over the C2 pads using the 4way interface. However, I seem unable to get BLHeli suite to recognize the ESC - while I believe I have everything wired in the correct way.

Some details:
ESC: Aikon 20A SEFM
Arduino: Nano V3.0
BLHeli Suite: BLHeliSuite16614903

I am powering the ESC with an external 12V power supply, it is showing a (low) current running after powering the ESC.

Regarding the pads: I checked (multimeter) that the pad near the ESC corner is the GND pad. I then followed the usual logic and assumed the furthest most pad is the G2D pad, the pad right next to it being the C2CK pad.

I know the ESC is working, it beeps nicely when powering it.

Some pictures regarding the situation:
[image]
[image]

Am I doing something obviously wrong here? Are the C2 pads on the Aikon 20A different than what I assume?

swilkens, 2017-07-29:
Additionally, BLHeli Suite logs mention that all interface lines are LOW.... however, measuring with my multi-meter shows a 3.3V level on C2D and C2CK

4712, 2017-07-29:
1. Please ensure to use the right hex file for the Arduino.
2. the C2 pad layout might have changed with the SEFM 20A (I have some here, where it is changed) compared to the 30A.
I guess you connect GND right, but D3 seems connected to VDD. 
GND
C2D or C2CK?
VDD
C2D or C2CK?
You might have a look here:
[image]

swilkens, 2017-07-29:
1. I used BLHeli Suite to create the linker, I used the MULTI hexfile (4wArduino_Nano__16_MULTIv16400.hex)

2. The picture may be vague, I made a new picture to clarify the connections:
[image]

I have also tried this with an Arduino MEGA and a Aikon 30A SEFM (v3.1), with the same result.

You mention C2 pads have changed, what changed on the ones you have?

4712, 2017-07-29:
Try your C2CK wire at the pad near GND. 
like: C2D - NC - C2CK - GND

If it doesn't connect exchange C2D with C2CK

swilkens, 2017-07-29:
Your very first suggestion worked on my 20A, it seems they didn't follow the standard pad lay-out on this 20A SEFM.

I will verify if the same thing applies to my 30A SEFM (V3.1). If so I will let you know here, maybe a note can be made in the relevant manual section for BLHeli.

swilkens, 2017-07-29:
@4712 

Turns out the 20A SEFM were just oddly padded, I was able to flash them over 4way like this:
`( D2 )  (  )  ( D3 )  ( GND )`

Which would work out to:
`( C2D )  (  )  ( C2CK )  ( GND )`

My SEFM 30A (V3.1) had standard padding, so it seems it was just the SEFM 20A that had alternative padding. Thanks for your time and suggestions, maybe something for the manual - even if we are at the dawn of BL32 and the dawn of BL_S.

Closing this :)

closed 2017-07-29 by swilkens