bitdump/BLHeli #252: Emax 40A with Unknown device error

juvinski, 2017-08-15:
Hi,

I received a Emax 40A but instead the others, the resistor and capacitor on the top right of the boards wasn't present, and when I tried to flash I received the Unknow device Error.
Anyone has flashed this esc ?
This is a pictures of the ESC.
https://drive.google.com/open?id=0B1omx8hPUswTM2p6Y0ZiSUJhcTA

lucasperin1, 2017-09-17:
opa, cara eu comprei esses emax 40 a , mas nao estão funcionando direito, vc poderia ajudar?

juvinski, 2017-09-17:
Opa beleza ?

Cara meu problema era o adaptador que eu acho que queimou, montei um arduino como adaptador e aí consegui atualizar o firmware, qual problema você está tendo ?

lucasperin1, 2017-09-18:
beleza, eu comprei um arduino também atualizei, quando eu calibro os esc na
pixhowk beleza funciona perfeito, depois que eu desligo e ligo, os motores
já armam girando forte e quando acelero o motor fica todo errado, já
atualizei os esc os meus já vieram sem aquele capacitor e resistência que
incomodavam , agora já não sei mais o que fazer, se tu souber de algo
agradeço.

juvinski, 2017-09-18:
Uma pergunta,

voce já tentou calibrar eles direto pelo receptor sem a controladora e como ficou o comportamento ligado somente no receptor ?

lucasperin1, 2017-09-18:
sim fica bom, a calibração na pixhowk fica boa também, mas na hora de armar
fera tudo

juvinski, 2017-09-18:
E você ja usou essa pixhawk antes? Calibrou o rádio direitinho ?

lucasperin1, 2017-09-19:
Não usei, faz tempo que comprei um ano quase eu comecei a montar, aí chegou
nessa parte não progredi mais, sim calibrei tudo pelo mission planer, acho
que vou comprar outros esc e abandonar esses, o que voce acha? agora
comprei um racer to esperando chegar, compro a maioria das coisas da china
ai demora um poco.

juvinski, 2017-09-19:
Você consegue me mandar um print da configuração que está no esc? Conectar ele com o blheli e capturar a tela ?

lucasperin1, 2017-09-19:
sim, quando chegar em casa eu faço isso te mando  dai, obrigado.

lucasperin1, 2017-09-19:
segue a  configuração em anexo .

juvinski, 2017-09-19:
Cara  não veio não.

lucasperin1, 2017-09-19:
> On Tue, Sep 19, 2017 at 7:07 PM, Vinicius Juvinski <

foi agora?

juvinski, 2017-09-19:
Veio nao,

coloca no google drive e posta o link ai :)

lucasperin1, 2017-09-19:
https://drive.google.com/open?id=0B8hfP0kc2ksDcGRPNFEwejB0eTQ

lucasperin1, 2017-09-19:
https://drive.google.com/open?id=0B8hfP0kc2ksDeGphWm85ZUNNeFU

juvinski, 2017-09-19:
Putz cara,

os meus 2 esc sao iguais esse seu ai e estão ok.
Estranho que se está conectado no receptor e funcionando e somente na pixhawk que não, parece algo na controladora. Qual software você está rodando ? plane, copter ?
Qual versão ?
Ta usando a safery switch ?
Pode mandar um print da tela de calibração do Radio ?

lucasperin1, 2017-09-19:
https://drive.google.com/open?id=0B8hfP0kc2ksDMWxOcXMxX3ozbk0

lucasperin1, 2017-09-19:
desativei a switch, copter to usando

2017-09-19 20:31 GMT-03:00 Lucas Perin <lucasperin.lp@gmail.com>:

juvinski, 2017-09-19:
Cara vc ta com um problema de configuração, saca só:

O radio na pixhawk ta com intervalo de 982 para 2006, já no ESC está 1012 e 2020.
O centro também está diferente.
Como você calibrou o ESC ?
A switch só vai fazer diferença para armar - e calibrar se for o caso.

lucasperin1, 2017-09-20:
Mas é pouca coisa os números 

Enviado do meu iPhone

lucasperin1, 2017-09-20:
Tu acha que essa pouca diferença interfere ?
Enviado do meu iPhone

juvinski, 2017-09-20:
Olha cara,

se está funcionando no radio e pela PixHawk não, pode ser uma tentativa, como você calibrou o ESC ? Eu normalmente plugo o esc no receptor direto - não uso calibrar pela controladora.

lucasperin1, 2017-09-20:
Pelo receptor funciona bem, como te falei no modo de calibraçao da pix funciona como no receptor, aí quando acabou a calibração eu desligo espero um pouco e ligo aí parece que perdeu a calibração, arma girando forte e o acelerador fica lento e sem escala.
      
Enviado do meu iPhone

juvinski, 2017-09-20:
Qual o receptor que você está utilizando e como ele está conectado na pixhawk ? Sbus ? PPM ?

lucasperin1, 2017-09-20:
Bom dia!  ontem a noite tive uns problemas não pude continuar, meu receptor
é um frsky x8r taligado na sbus.