bitdump/BLHeli #254: Add EMAX Simon Series 30A to supported ESCs

Kaeltis, 2017-08-24:
After first thinking these ESCs (https://www.emaxmodel.com/esc/simonk-series/emax-simon-series-30a-for-muti-copter.html) would not work with BLHeli (since they are not listed in the PDF), I got in contact with the EMAX support.
They told me to flash "EMAX_20A_MULTI_REV14_9.HEX" on the ESCs via the 4way-Interface and it worked perfectly.

I couldn't test the ESCs on a copter yet, but the flashing went through without any issues.

Here is a Picture of the pads they sent me:
[image]

benjinne, 2018-07-22:
I can't seem to read the setup when I connect with a nano using BLHeliSuite 16.7.14.9.0.1. Can you explain how did you managed to flash this?

Kaeltis, 2018-07-23:
Yup, still have these on my drone and haven't changed anything since 2017, but also didn't get more than a couple of hours flight time on them, so can't say anything about the long-term (should be fine though).

benjinne, 2018-07-24:
I just flew with the "EMAX_20A_MULTI_REV14_9.HEX" and it flew great! I flew more than an hour and had no problems.

Kaeltis, 2018-07-24:
@svmdevelopment the firmware is "EMAX_20A_MULTI_REV14_9.HEX", you can just follow the official flashing instructions for the flasher you have, using the 3 pads shown in the picture in my first post.

If you want to change settings over the 1-wire signal cable of the ESC, you have to additionally remove the two parts marked in the picture:
[image]

JLVALENCIA-HUB, 2020-05-07:
Good Afternoon,
I have just joined the  community, thank you very much for accepting me. I came accross this thread when looking for information about which file to flash to my 4 blheli Emax ESCs 30A. I rise the question to Emax Support and the told me the following. I know this thread is old, but I thought it might help some one with the same problem I've been struggling with for weeks. In a nutshell, the programm is the same for either models, the 20A and the 30A. Boths ESCs are only different regarding hardware.
[image]

alexvazquez, 2020-10-17:
I have the 30A version and I just did the same and Flash the 20A and works fine. The only problem is that the LED on the board wont work anymore. At least it happened to me.