bitdump/BLHeli #258: BLHeli32: bidirectional mode broken

leisim, 2017-09-17:
I tried using the Spedix GS30 ESCs for flying 3D. After arming everything seems fine but once the throttle is slightly negative, the motors stop spinning and make strange noises.
The ESCs have to be disconnected from power to stop the twitching

Both, bidirectional mode and bidirectional_reverse mode have this problem.

mikeller, 2017-09-17:
I can confirm the same behaviour for Wraith32, version `Multi_32_12_Dshotprog_Test_Code`.

sskaug, 2017-09-17:
Yes, the bidirectional performance of BLHeli_32 has not been reliable so far.
There is a rev32.2 code coming that has very good bidirectional performance - I've flown tic-tocs with it myself :). I've posted test codes (rev32.15) on github. Please test!

leisim, 2017-09-17:
Thanks for the fast response ;)
I've flashed the firmware but I don't get the 5th beep at the ESC startup. And the quad does not arm. What am I missing?

sskaug, 2017-09-17:
Sorry for this. I used Multishot when testing, and there was a silly bug in Dshot signal handling for bidirectional. Please try again, fixed code is posted.

leisim, 2017-09-17:
Works great. Thank you :)

closed 2017-09-17 by leisim