bitdump/BLHeli #259: Dshot programming not working with BLHeli32 32.15

mikeller, 2017-09-20:
Changing ESC settings with Dshot commands seems to be broken in 32.15. See the Betaflight CLI output  from trying to disable 3D: ESC 0 running 32.15 won't change the setting (I tried multiple times), whereas ESC 1 (flashed back to 32.12) changes on first attempt. One more thing I noticed is that, when receiving a Dshot command, the motor on the 32.15 ESC twitches, whereas the 32.12 ESC doesn't.

```
# dshotprog 255 6
Using all outputs.
Info for ESC 0:
ESC: #Airbot_Wraith32_ST#            
MCU: 0x
Firmware: 32.12
Rotation: normal
3D: on
Low voltage limit: 3.0
Current limit: 40
LED 0: on
LED 1: off
LED 2: off
LED 3: unsupported
Info for ESC 1:
ESC: #Airbot_Wraith32_ST#            
MCU: 0x
Firmware: 32.15
Rotation: normal
3D: on
Low voltage limit: 3.0
Current limit: 40
LED 0: off
LED 1: on
LED 2: off
LED 3: unsupported
Command 6 written.

# dshotprog 0 9 12
Using output 0.
Command 9 written.
Command 12 written.

# dshotprog 1 9 12
Using output 1.
Command 9 written.
Command 12 written.

# dshotprog 255 6
Using all outputs.
Info for ESC 0:
ESC: #Airbot_Wraith32_ST#            
MCU: 0x
Firmware: 32.12
Rotation: normal
3D: off
Low voltage limit: 3.0
Current limit: 40
LED 0: on
LED 1: off
LED 2: off
LED 3: unsupported
Info for ESC 1:
ESC: #Airbot_Wraith32_ST#            
MCU: 0x
Firmware: 32.15
Rotation: normal
3D: on
Low voltage limit: 3.0
Current limit: 40
LED 0: off
LED 1: on
LED 2: off
LED 3: unsupported
Command 6 written.
```

sskaug, 2017-09-20:
Thanks for testing and reporting!
This is a result of dshotprog functionality being merged with bidirectional mode improvements.
32.12 had only dshotprog functionality, while 32.15 and the released 32.2 have both.

What happens, is that in bidirectional mode, a dshotprog command also results in a short throttle pulse which spins the motor a few revolutions. Therefore "dshotprog 0 9 12" does not work, as the "9" results in nonzero throttle that causes the "12" command to be missed. On the other hand, "dshotprog 0 9" followed by "dshotprog 0 12" works.

This is only an issue when doing dshotprog commands in bidirectional mode.
Motor spins a little but not in any hazardous way.

A fix will be implemented on the next rev, or on beta codes upon request.

mikeller, 2017-09-20:
Thanks for the explanation. I think it's only an edge case that is broken,  so as long as there is no safety implication I agree that fixing it can wait until the next release. I reverted my test set up back to 32.12 for Dshot testing.