bitdump/BLHeli #261: BLHeli32 - Siskin_ST 32.2 version errors

tarekmaalouf, 2017-10-05:
Running DSHOT 1200 the Siskin_ST 32.2 update that just released seems to cause a multitude of issues. I have experienced the following:

1. Overheating rather quickly - within seconds and too hot to touch
2. Motors will not spin up properly - they will attempt to spin and then the entire configuration is gone and BLHeli only recognizes 1 ESC. 

Reverting back to 32.1 takes things back to normal. What can be done about this?

sskaug, 2017-10-06:
I just tried this, using the released Rev32.2 Siskin_ST code and a REVO with 3.2 RC4 code, and I can not reproduce any problems with Dshot1200 using the motors tab. Any special conditions or settings in your setup?

tarekmaalouf, 2017-10-06:
I will put a quick video with it so that I can show you what I am experiencing. I will post back here shortly.

4712, 2017-10-06:
I checked it here also with 4s using the  BLHeliSuite motor tab and "Special Tests" feature. I did not find any issue.
A video would be fine, but might not show the way how to reproduce what you are experiencing.
Listing some items like: specific FC, BF/CF version, FC configuration, motor, Lipo... could be helpful.

tarekmaalouf, 2017-10-06:
Ok here is what I have for now to show:

FC - Lumenier F4AIO - https://www.getfpv.com/lumenier-f4-aio-flight-controller.html
ESC - Lumenier BLHeli 32bit 4-in-1 - https://www.getfpv.com/lumenier-blheli-32-35a-4-in-1-esc-2-4s-w-bec-3a-12v-1a-5v-dshot-1200.html
Motors - Lumenier - https://www.getfpv.com/lumenier-mb2206-9-2450kv-freybott-motor.html
Battery - Tattu r 4s - https://www.getfpv.com/tattu-r-line-1550mah-4s-95c-lipo-battery.html

Ok so here is the deal - issues are not just limited to the ESC but this particular thread specified ESC because I am able to duplicate this each time when I change from 32.1 to 32.2 

The FC is running LUXF4OSD as recommended from Lumenier when using BF 3.2.*. This in itself has caused issues with the FC in regards to acc calibration but I have found a work around by flashing with the BF 3.1.* recommended firmware GOODDHF4 and then flashing back to the LUX version.

In regards to the ESC, the issue is duplicated (not necessarily the overheating) each time I upgrade. The firmware upgrade processes perfectly but once it is complete, if I attempt to start the motors I will get an initial spin up (very low RPM assume around 1050 or so on throttle) and then it all stops. At that point when I attempt to reconnect to BLHeli only one ESC will show up - #3 in most cases. The ESC will not initialize from that point forward. I have to remove the Lipo and the USB and then reconnect and then Flash back to 32.1 to get response. 

In the case with the overheating, the motors were not spinning. They, along with the ESC, became extremely hot. I finally unplugged the items, let them cool, flashed down to 32.1 and the quad flew like a champ. 

If there are any screen shots I can give you please let me know. I am prepared to provide as much as I can at this point.

One question I do have..just for curiosity, since the DSHOT1200 does not require motor calibration, is there a chance that the ESC can be damaged or otherwise programmed incorrectly if a calibration is attempted via BF? Through testing I wanted to see the response so that if my clients do this I can issue a warning or such - and I noticed that while the beeps to not complete the full initialization to show the motors are in calibration mode, once I bring down the throttle in BF I do hear the final beeps as if calibration ended. I know this is off the topic of the original post but I thought I would ask just to see if accidental calibration attempts on DSHOT1200 ESC's would risk damage to the board. 

Thank you very much for your help.

sskaug, 2017-10-17:
Fixed in Rev32.3

closed 2017-10-17 by sskaug

tarekmaalouf, 2017-10-17:
going to test it tomorrow first thing! Thank you