bitdump/BLHeli #262: Programming for BF (compiling etc)

tarekmaalouf, 2017-10-06:
While this is not really an issue it is more of a question presented to many of you...

I am a computer engineer specializing in security and data protection but have also put many years into programming. I have a total of roughly 32 years in the industry, with programming dating back into the dinosaur ages. I want to provide my resources to the BF project and wanted to know if anyone can tell me how to get started (if even possible). Simply put, with the resources at my disposal, I feel as though I could really give back and would like the chance to do so. If it is not possible then I understand..just wanted to reach out and see.

4712, 2017-10-06:
I guess, this is what you are looking for: https://github.com/betaflight

mikeller, 2017-10-06:
Hi @tarekmaalouf. Probably a good starting point to get into the Betaflight project is to join the Betaflight Slack at http://www.betaflight.ch/ . Most of the core developers are in there, and you can discuss your ideas and plans.

tarekmaalouf, 2017-10-06:
thank you both. I will do just that...appreciate your time..I will close this now!

closed 2017-10-06 by tarekmaalouf